samtap/fang-hacks #196: Can't get access to http://device-ip/cgi-bin/status

dzalif, 2017-07-21:
Sorry, I am newbie. I buy this xiaofang yesterday but I can't get access to http://device-ip/cgi-bin/status yet. and It's not displaying 404 too. I have install the hacks to sd card, connect the camera to my wifi hotspot, and try to access from my computer browser but its display nothing. Anyone can know this problem ? Sorry for my english.

xwarior2017, 2017-07-22:
firmware 3.2.0.30? 

wait for next update :D

dzalif, 2017-07-22:
finally, the hack works. The video running wells on  Pc. But can I play rtsp stream on mobile ? I try with VLC android  but failed. I need it for my android app project.  anyone know the problems ?

samtap, 2017-07-22:
Try tinyCam but it should work in any rtsp client

dvbit, 2017-07-24:
I join the chorus :-( received my cam.
registerd in mihome.
has it's ip 192.168.2.107 (can ping it and see it on router) 
when i put in the sd I hear the clank 
no access to status page though (404)
hello.cgi responds with blank page.

Further info: Camera records on sd ! If i put the sd in a pc I can find recordings
Camera updated to latest firmware
mihome version .36

Any suggestions?

Thanks for the good work on the nice gadget

dvbit, 2017-07-24:
ssh connection refused
first attempt at status page gave connection refused then 404

dvbit, 2017-07-24:
@dzalif what have you done to make it work?

dzalif, 2017-07-24:
First I install Mi home on my phone, the blue led always blink and stuck on loading 1/3. And I try with another phone with new Mi account and follow the step untill the app connect to the camera via remote connection, the blue led is solid now. I try put the hacks again, and then works.

dvbit, 2017-07-25:
Just a short update.
Succesfully applied.
Camera is MAC34
MiHome latest
Problem was with latest fw update.
Downgrade and it will work.
Double check autoupdate plugin in MiHome as it can trick you and do it... (I got caught the first time)

varoufitsa, 2017-07-25:
@dvbit 
Can you write down the steps you followed in order to downgrade?
I'm trying since yesterday without success.

furmek, 2017-07-25:
@varoufitsa I had the same problem and also doing a downgrade helped
I followed the steps listed on [the wiki](https://github.com/samtap/fang-hacks/wiki/HowTo:-Flash-original-Xiaomi-firmware-from-sdcard-(factory-reset)) the SD Card section.
And I've also disabled internet so the camera wouldn't update itself.

desaintmartin, 2017-08-04:
I had the same problem. Fixed it by throwing away my crappy SD card and buying a new one.

alsnow22, 2017-08-07:
how did you get it work I have firmware 3.2.0.30?

aleifuu, 2017-08-17:
Hi all, 

first of all, great github project to liberate these small devices. kudos sir !

however, I too run into a problem of applying the hacks since I get a 404 on /cgi-bin/status and OpenDir Failed on /cgi-bin/hello.cgi, so I guess the autorun script is not running 

Both my xiaofang are running fw 3.0.3.56 so it should not be a problem

I have followed the steps outlined in README as well. wait untill the blue led becomes solid then insert the already flashed sdcard. I even try another card as well but getting same result

I downloaded the 2nd version fanghacks as in the release notes, and dd it to the card ( I was on Mac )

Yes, I don't get the Ding-Ding sound when inserting the card. So that means the hack sort of 'failed' to be recognized, is it ? 

Maybe someone could share a light on this ?

samtap, 2017-08-17:
@aleifuu Do you have the cam paired up with mi home?

aleifuu, 2017-08-17:
Hi @samtap, Yes I have it paired up with Mi Home. I can access to it from Mi Home no problem as well

aleifuu, 2017-08-17:
Hey, I just got home and try again with a different empty microSD card. I've got Ding-Ding sound. hmm, let me flash fanghacks.img to this card ... lol 3rd time lucky maybe

aleifuu, 2017-08-17:
just little update, it still doesn't work

I'm not sure if they are something wrong with the way I dd it. but seems fine otherwise

sudo dd if=fanghacks_v0.2.0.img of=/dev/disk1s1

514048+0 records in
514048+0 records out
263192576 bytes transferred in 808.086525 secs (325699 bytes/sec)

maybe I should try earlier .img version ?

samtap, 2017-08-17:
@aleifuu The image contains a full disk image with partitions, so it must be written to a disk not to a partition. If the sdcard on your Mac is /dev/disk1, the command would be dd if=fanghacks_v0.2.0.img of=/dev/disk1 bs=4M

aleifuu, 2017-08-17:
Hmm, yeah that's what I thought I was doing. But I think you are right

I'm flashing the .img now on raspbian, something that I'm more used to.
Flashing to a disk instead a partition

I'll post a little update later on

aleifuu, 2017-08-17:
ok some update

/hello.cgi is now returning

bootstrap

snx_autorun.sh

time_lapse

record

log_34CE00F0665A_s.txt

/status

still 404 however 

I created 2 partitions : vfat and ext2 as in README

I think it's still something to do with the somewhat not correct partition table ? 

I can connect to it using MiHome app, it's all fine. just can't get to the status coz it's 404 LOL

aleifuu, 2017-08-21:
Try again with same card, but flashing it using Win32 Disk Imager instead. It works !

---
Applying fang hacks scripts...
Running fang_hacks.sh...
Removed stale logfile
Executing script (enabled: 1)
Waiting for cloud apps...
iCamera is running!
Boa webserver is running!
Not linking /media/mmcblk0p1/bootstrap/www/action: already exists
Not linking /media/mmcblk0p1/bootstrap/www/func.cgi: already exists
Not linking /media/mmcblk0p1/bootstrap/www/network: already exists
Not linking /media/mmcblk0p1/bootstrap/www/scripts: already exists
Not linking /media/mmcblk0p1/bootstrap/www/status: already exists
Failed to find hacks in /media/mmcblk0p2/data!
Mounted /media/mmcblk0p2/data
Running startup scripts
Stopping Cloud applications...
  test_UP[630]
  iCamera[628 619]
  miio_client[815]
Module snx_wdt unloaded (0)
Starting Network...
Cloud mode: network unmanaged
Starting ntpd...
Starting telnetd on port 2323...
Starting dropbear on port 22...
Starting FTPd...
Starting RTSP server...
Starting IR script...
Finished
---

just a case of wasn't properly flashing it I guess. Great work !

aleifuu, 2017-08-21:
Now that I get the hack to work. I've got some questions that 

a. ) "Mi Home" app will no longer work for you to view your xiaofang camera. hence you'd want download an app like "TinyCam" to view your xiao fang, is that right ?

b. ) Question of disabling cloud 

I don't want to send streaming data to a cloud that is unknown to me. I'd just like to do it old-school when I just vpn to my home network and see what camera is seeing without first sending it to unknown cloud. How can I achieve this if I don't want to DISABLE THE CLOUD APP permanently ?

fubar2, 2017-08-21:
@aleifuu on b) if your router is configured to prevent any internet access for the camera then it cannot _phone home_ - some cloud processes may still be running but at least they can't do much.

samtap, 2017-08-21:
@aleifuu Set network to client mode, make sure it works with wpa_supplicant.conf instead of the cloud credentials (reboot), then enable the "disable cloud" checkbox on the status page. If you want to re-enable it just uncheck that box and reboot the cam. You could even make a script to start the app without rebooting (have it run /etc/app/iSC3S) but you'll have to stop rtsp server first.

kenavy, 2017-08-25:
Hello,
I just got my camera. I installed the image on an SD card. I hear the "ding ding" but the page http://device-ip/cgi-bin/status return 404.

I do not know what else to do

rushtamas, 2017-09-08:
The hack won't work for me until today when I soldered the rx tx gnd connectors to a usb - serial converter to see what happening exactly. 
I logged in via serial  console (un: root, pw: ismart12), started telnetd.
So I can login via telnet, and now I see the problem was with the SD card partition names.

If the hack not works, maybe you need to use linux or macos to correctly partition SD card,
not win32diskimager!

samtap, 2017-09-09:
@rushtamas There's no problem with partition names that I know of. You don't need to partition the sdcard, it is all included in the image, regardless what you use to write it to the card.

Swaktor, 2017-10-08:
DOWNGRADING to 3.0.3.56 from 3.2.0.30  And Installing Fang Hack without Error 404. !!  Solution !!
________________________________________________________________________________________________________

I found this Tutorial Custom firmware  in English Language.
https://www.youtube.com/watch?v=mpzPWYONWZA

I just put this .bin File on my FAT32 formatted 16GB MicroSD Card with only 1 Volume / Partition and downgraded it as it is shown in the Video.

After downgrade, connect your XF Camera with the MiHome App 4.0.11 or 4.0.11 to the Wifi- I prefer using .11 

Then I deleted the .bin File and copied the Fang hack Master File into your root folder of the MicroSD Card. 
Get it here  and dowload the zip File : https://github.com/samtap/fang-hacks

Insert the MicroSD card while you XF Cam is Connected to the WiFi and then type 
http://camera.ip.address.here/cgi-bin/status. Thats it ! 
For me there was no Error 404 and I was able to check the http://camera.ip.address.here/cgi-bin/hello.cgi