samtap/fang-hacks #197: Help, Is there a way to play Rtsp stream with VLC android ?

dzalif, 2017-07-24:
I don't know why the rtsp stream for this camera didn't work on VLC android, I use VLC android to stream for the webcam and works, I did it too with tinycam, but why it doesn't work on VLC android ? anyone know the reason and how to solve it ?

dvbit, 2017-07-24:
BTW there's a strange thing : the fw of the camera is reported as .56 
I do not know what actually happened (if I downgraded or not) however the downgrade procedure of the wiki lead to have the problem solved 👍

dvbit, 2017-07-25:
I have the same issue. Also on TinyCam Pro on android video is choppy (a few frames and then black) any clue?

samtap, 2017-07-25:
Probably wifi bandwith is too limited. Try changing rtsp server settings (lower resolution and/or bitrate)

dzalif, 2017-07-26:
I have change the resolution to 360p. How do I change the bitrate ?

samtap, 2017-07-27:
Use the -b option: https://github.com/samtap/fang-hacks/wiki/snx_rtsp_server-help-text

dzalif, 2017-07-27:
Thanks 😃 . That note help me a lot.

closed 2017-08-18 by samtap

homemadev, 2017-11-26:
I'm having this issue now. Changed my `/media/mmcblk0p2/data/etc/scripts/20-rtsp-server` start line: 

from `snx_rtsp_server -W 1920 -H 1080 -Q 10 -b 4096 -a`
to `snx_rtsp_server -W 640 -H 360 -Q 5 -b 512 -a`

but VLC still crashes on my android. Works great from desktop. 

I'm not saying it's an issue with fang-hacks. @dzalif or @dvbit would you post your config that works with VLC on android?

dzalif, 2017-11-26:
It doesn't work on VLC android. Try VXG.

homemadev, 2017-11-27:
Perfect. I get audio and high resolution video on VXG. Thank you!