samtap/fang-hacks #198: Rotate RTSP stream

gokuu, 2017-07-25:
Hello!

Is it possible to rotate the RTSP stream by 180º? I had to mount the camera upside down, and would like to rotate the stream, if possible, before it actually leaves the camera.

LouiS22, 2017-07-25:
Hello there,

Just follow Bobby Romero's guide [http://bobbyromeo.com/technology/xiaomi-smart-1080p-wifi-ip-camera-rtsp-streaming-hack/](url), and you'll get there.

You need the "Install snx_isp_ctl" section, and "Flip/Mirror Image" section. It's like a 2 minutes job, did it yesterday myself the same.

gokuu, 2017-07-25:
Perfect, thanks!

closed 2017-07-25 by gokuu

kszere, 2018-04-07:
Link not available.

Bechrissed, 2020-08-16:
He (Bobby Romero) modified urls, the article is still there. Now to found under the following link: http://bobbyromeo.com/home-automation/xiaomi-smart-1080p-wifi-ip-camera-rtsp-streaming-hack/