samtap/fang-hacks #199: Source file not found: '/media/mmcblk0p1/bootstrap/fang_hacks.sh'

garylau, 2017-07-25:
Hi there, I download the 0.2 img file and write it to my SD card. I assume I can't explore the 2nd partition in windows anyway so I didn't found the data folder.

After the blue led stopped blinking, I insert the SD card and heard the hammer sound.
I was able to browse the status page. However after I clicked 'apply', I got this error message straight away.

> Applying fang hacks scripts...
> Source file not found: '/media/mmcblk0p1/bootstrap/sdcard'
> Source file not found: '/media/mmcblk0p1/bootstrap/rc.local'
> Source file not found: '/media/mmcblk0p1/bootstrap/rcS'
> Source file not found: '/media/mmcblk0p1/bootstrap/fang_hacks.sh'
> Source file not found: '/media/mmcblk0p1/bootstrap/fang_hacks.cfg'

Did I miss anything? Thanks for your help!

garylau, 2017-07-25:
Just checked the firmware version. It is still 3.0.4.9 (it's asking me to upgrade to 3.2.0.30 though)

samtap, 2017-07-25:
Looks like it can't find some files on the first partition of the sdcard, that one is accessible by Windows so you could check if the files are there (likely the card is defective/corrupt).

nehaleem, 2017-08-04:
I have same issue. I tried to put `sleep 30` into `snx_autorun.sh` but its not working. Thing is, when camera boots, status page will result in 404. But after reinsert the sd-card status its working, but mount point is different i think and thats the issue with source file not found after click on Apply.

nehaleem, 2017-08-04:
OK, i found a solution and all my cameras are working now :-) Yeey

1) Format SDCard like in wiki (vfat + ext2)
2) Download this source code + i dont want to use Mi Home app, so i will download this too and copy it to rest of the source code https://github.com/samtap/fang-hacks/issues/132, edit both dot files (wifi password and ESSID)
3) Copy bootstrap folder, snx_autorun.sh, wlansetup.sh, .wifipasswd, .wifissid to vfat partition
4) Copy data folder to ext2 partition
5) Put SDCard in camera, power on camera. Camera will start blink blue after few seconds (if wifi connection is ok)
6) Power off camera and REMOVE sdcard
7) Power on camera without sdcard, wait to wifi connection (now its configured) till blue light blinks
8) After blue that, put sdcard inside camera, wait till sound "klink klink". Go to the `http://ip/cgi-bin/status` and click Apply.

Thats all!

samtap, 2017-08-18:
You could go through all this effort, or just pair your camera with mi home to setup the wifi connection. Anyway I'm happy your issue is resolved, so I'll close this.

closed 2017-08-18 by samtap

jia175891641, 2018-08-16:
我也解决这个问题了   不用这么麻烦  用树莓派官网的win32 disk imager就ok了 之前用的是其他版本的win32 disk imager