samtap/fang-hacks #201: IR constantly toggling when dark

andybotting, 2017-08-10:
I've just received my XiaoFang device (34 MAC), and installed the hacks successfully.

As soon as I tested it in the dark, the IR mode constantly switches on and off every three seconds with audible click of the filter changing. 

Picture looks fine, and if I quickly disable the IR script at the right moment, I can leave it in IR mode and all is well in the dark.

I've tested the original version of the IR script, and the updated one for the heat fix.

Any ideas?

andybotting, 2017-08-10:
I'm running v3.0.3.56, which was the firmware it came with.

I can see that when as soon as 

`gpio_aud write 1 0 1    # enable ir led and latch the filter in the correct position`

is called, the value of

`DAY="$(gpio_aud read 2)"`

is always 1, so the IR will simply shut off again. 

Would have been useful to test this before applying the hacks.

andybotting, 2017-08-10:
I've taken the SD-card out and replicated the issue using the MiHome app. Is this enough to confirm whether this is a hardware/sensor fault?

samtap, 2017-08-18:
Perhaps the LDR is defective? The 'gpio_aud read 2' value should change if you cover the sensor or when you shine a light on it. If it doesn't, the LDR sensor is probably defective.

andybotting, 2017-08-19:
@samtap Thanks for the reply. I could replicate the issue with the Mi Home app too, so I suspect it was defective and I've sent it back as a warranty job. Should have a new one arriving this week - I'm sure it'll work better.

closed 2017-08-19 by andybotting

andybotting, 2017-08-25:
New camera arrived and (as expected) all good.