samtap/fang-hacks #202: Update RTSP server to h264_v4l2_rtspserver

gokuu, 2017-08-11:
Seeing as the RTSP server project hasn't been updated for 2 years now, and it is also based on [this one](https://github.com/mpromonet/h264_v4l2_rtspserver), could it be possible to update the RTSP server to this one? It supports RTSP over HTTP out of the box, and also protect the RTSP stream with a username and password...

Or, at least, could you give me some pointers on how I could go about compiling it to work on the camera? :)

samtap, 2017-08-11:
Yes this has been mentioned many times, expect updates soon

closed 2017-08-11 by samtap

snoerenberg, 2017-08-11:
@gokuu @samtap this would make it easier to get it to work with Zoneminder in a Docker container. I had trouble (couldn't get it to work) with the UDP traffice into my container. Within Hyper-V directly I could get it to work.

samtap, 2017-08-11:
The new server only supports TCP