samtap/fang-hacks #203: Any way to power off USB at the back?

charlesmigrino, 2017-08-14:
I am attaching a mini USB fan at the back to cool down the camera. Night vision can get a little crappy when the devices heats up and this improves it by a lot. I'd like to power off the USB when it is not using the IR lights which can be controlled via the 21-ir-control script. Is this possible?

charlesmigrino, 2017-08-14:
By the way, I'm using Xiao Fang. :)

samtap, 2017-08-18:
Not possible without heavy hardware modifications. It should not get so hot that it needs a fan anyway, are you using the updated ir script?

charlesmigrino, 2017-08-22:
@samtap Yes, I am using the updated IR script. I am really not worried about the camera getting hot, it's the night vision quality I want to improve. I have noticed that as the camera goes longer in IR mode, the quality deteriorates. I found a forum somewhere (I can't remember where) that the heat is actually to blame and the script update is not sufficient even though that has reduced the temperature. I have tried adding a USB fan on the camera and it really did improve the quality though I wasn't really sure enough back then since the fan had a LED on it which kinda brighten up the room. Recently though, the LED got busted so I can 100% prove it now that cooling it down can help. By the way I live in a tropical country so not all people would find this relevant. Comparison of camera without fan and with fan below.

[image]
[image]

charlesmigrino, 2017-08-22:
The main reason why I wanted to power off the USB at the back was fully due to the LED of the fan. I didn't like it turned on during the day when the IR is off. But since it got busted then I don't really need it anymore but it would be really nice to have the option though, if it was possible, hardware-wise.