samtap/fang-hacks #209: Apostrophe in WiFi SSID - Client Mode does not work

adrianmace, 2017-08-21:
Running the most up to date firmware that works, I cannot connect to my SSID titled **Adrian's Family** without the device breaking to the point where I need to reboot into cloud mode and potentially reset the device to regain connectivity.

samtap, 2017-08-21:
It's a known issue, easiest is to manually enter the SSID in /media/mmcblk0p2/data/etc/wpa_supplicant.conf
Or check the update to the network cgi script, it's a very simple fix: https://github.com/samtap/fang-hacks/commit/f3c3fcdf99f2e1ec520e911d6d31149a1075e8c9

adrianmace, 2017-08-21:
Thank you for the tip. I seemed to jump the gun a bit because I did find and modify wpa_supplicant.conf myself and it worked correctly from there. Thanks for the assist.

Please note it also required manually editing the elements in the HTML form to move the variables around and modify the spans so that it included the correct info, then un-disable the apply button and apply the changes without testing first.

I guess in hindsight I could have just changed the /etc/*.config file and wpa_supplicant.conf but this was more fun :P

nimdy2017, 2017-10-24:
Is there also a problem with brackets in SSID's? I have (2.4G) at the end of my SSID to denote my 2.4Ghz and 5Ghz networks. I can connect the camera to my mobile hotspot in both Fang Hacks and the cloud app, but cannot get it to connect to my home WiFi.