samtap/fang-hacks #213: Cam is not working, just static orange led

pushpindersohi, 2017-08-24:
Hi, I got this cam couple of weeks back and i applied your hack through sd card. Everything was working fine. But one day, i was checking different options and accidentally i hit resize sd card. after that, cam rebooted and never came back. There's static orange light. Tried to re-flash firmware through sd card but no success. I also restarted it by inserting card first and kept it like that for 2 days but still light is orange and i am unable to login or get any info.

Is the cam bricked? is there any other way to get it working? I followed all your steps mentioned but no success yet. Please help.

Thanks.

samtap, 2017-08-24:
Does it still run cloud apps when you boot without sdcard?

pushpindersohi, 2017-08-24:
When i boot without SD card, nothing works. There is static orange light.
When i press setup button nothing happens. No blinking of lights or no
voice commands. I think its stuck in boot process. Any idea how to recover
it back?

*Pushpinder Sohi* | IT Specialist
345 Hudson Street, 6th Floor Suite 6A New York, NY 10014

[image:
http://static.wixstatic.com/media/d23af5_b4d7ec8a18b14ec48e704321e22a743f.gif_srz_371_296_85_22_0.50_1.20_0.00_gif_srz]

samtap, 2017-08-24:
I guess you disabled cloud apps? If the sdcard is corrupted by the resize (you rebooted before it was finished), it will refuse to mount hence not run any scripts to enable wifi etc. You said you re-flashed the image, which would get rid of any filesystem corruption, but also your custom wifi settings. If you have access to a Linux machine (or anything that supports ext2 filesystem) you can change the settings in data/etc/wpa_supplicant.conf. If you can't manage that, please check recovery options using the tmpl files on the fat32 bootstrap partition (which is accesible on Windows) described here: https://github.com/samtap/fang-hacks/wiki/F.A.Q#difference-between-stop-cloud-script-and-disable-cloud-checkbox (or search old issues, it's been discussed often)

pushpindersohi, 2017-08-24:
> fang-hacks/wiki/F.A.Q#difference-between-stop-cloud-
> script-and-disable-cloud-checkbox (or search old issues, it's been

I rebooted the cam before it finishes without realizing that its still
finishing things up. I dont have linux machine, but i tried recovery option
using tmpl files but doesnt work. I formatted sd card again and then tried
downgrading the firmware. Still nothing works.

So should i try with new SD card?will that resolve the issue of booting?

*Pushpinder Sohi* | IT Specialist
345 Hudson Street, 6th Floor Suite 6A New York, NY 10014

[image:
http://static.wixstatic.com/media/d23af5_b4d7ec8a18b14ec48e704321e22a743f.gif_srz_371_296_85_22_0.50_1.20_0.00_gif_srz]

samtap, 2017-08-24:
When you start doing random shit like downgrading and formatting you probably only digging a deeper hole for yourself. 
1. Flash the image file. It includes everything you need, you don't need to format or partition the sdcard, ever.
2. Carefully read recovery instructions. If it doesn't work, you're doing it wrong.

pushpindersohi, 2017-08-24:
I format card just to be on safer side if something goes wrong. Then i
flashed image file and followed the recovery steps. After placing recovery
file on sd card, i insert card in cam, then powerup the device by holding
setup button. I continued to hold setup button for couple of minutes but
nothing comes up. Orange light was still static even after 10 mins.

Any other suggestions or steps i need to follow. Really appreciated all
your effort and help on this.

Thanks.

samtap, 2017-08-24:
Read the FAQ and follow instructions.

closed 2017-08-24 by samtap