samtap/fang-hacks #215: Version 3.2.0.30 does not work with hack (for me anyway)

Gaunt1et, 2017-08-27:
All three of my cameras had to be downgraded for the hack to work. I would hear the sound but get 404 on the web page. [https://github.com/samtap/fang-hacks/wiki/HowTo:-Flash-original-Xiaomi-firmware-from-sdcard-(factory-reset)#via-sdcard](url)
Kudos to Yudi on the Discord.

Note: When downgrading insert the SD Card THEN power on WHILST holding the button for at least ten seconds. It takes a couple of minutes (in my case) for the downgrade to complete.

shortie84, 2017-08-29:
Hey Gauntlet, 

I've been trying this mutiple times but no luck. Seems you have had better luck so trying to see what I've done wrong.

Created a FAT partition called mmcblk0p1
Then copied the files over and renamed "0.elf" to "FIRMWARE_660R.bin"
Inserted the SD card then powered on the device and holding the Setup button at the same time for at least 10 secs.
Light blinks blue to orange for about 30 seconds then is now blue. How do I know when the downgrade is complete?

Gaunt1et, 2017-08-29:
Well I used Win32 Disk Imager to flash the fanghack image originally. It didn't work so I just deleted the files in those partitions and put the downgrade in the first partition where the hack used to be. Not really the way to do it I suppose but it seemed to work for me. Seems like you're doing it correctly, that's the only difference between my method and yours.

Verbank, 2017-09-14:
can this be done to a 34 device ?

samtap, 2017-09-15:
Yes it works the same on a 34 mac device. The led is static orange during firmware flash. After the downgrade when the cloud app starts for the first time it does some initialization of miio stuff and reboots.