samtap/fang-hacks #218: Will you good friends, hackers after the camera how to set the screen flip it?

1043717432, 2017-08-31:
Will you good friends, hackers after the camera how to set the screen flip it?
I need to hang the camera.

Chrischi-, 2017-09-01:
https://github.com/samtap/fang-hacks/wiki/Controlling-the-text-overlay

         --mfset-mode                   Set Mirror Flip Mode(mode:off:0, flip:1, mirror:2, flip&mirror:3)

1043717432, 2017-09-03:
@Chrischi- Thank you!

closed 2017-09-03 by samtap

1043717432, 2017-09-03:
Where is it set up?

Chrischi-, 2017-09-03:
telnet "ip of cam" 2323

U: root
P: ismart12

camhostname$ ./snx_isp_ctl --mfset-mode 3
Set Mirror Flip Mode(mode:off:0, flip:1, mirror:2, flip&mirror:3)

1043717432, 2017-09-08:
@Chrischi-  
Thank you, bother, I failed to log in tenlet login, can tell me how to operate?
[image]

U: root
P: ismart12

camhostname$ ./snx_isp_ctl --mfset-mode 3
Set Mirror Flip Mode(mode:off:0, flip:1, mirror:2, flip&mirror:3)