samtap/fang-hacks #219: Simple question

takinitease, 2017-08-31:
I bought two of these.   I'm not network or programming savvy.  I just need some help understanding on whether or not I can make this camera work here in the us.  I understand there is a hack, but it seems to have met mixed reviews of working.   With the hack, it works on tinycam?  What about remote viewing ? (Not being on Wi-Fi)...  I'm pretty need to the networking and ip camera arena do I'm just looking for some straightforward answers about this camera and whether or not I can get it to work...   thanks.  I definitely want to be able to monitor it from outside my home if possible. Otherwise I will just return.

Chrischi-, 2017-09-01:
You have to Port forward, to access the cam.

Use the Discord Chan to discuss this. This is **no** Issue.

closed 2017-09-01 by samtap

takinitease, 2017-09-01:
This is no issue? The fuck does that mean

Chrischi-, 2017-09-03:
For questions, suggestions or just general discussion, please join #fanghacks on irc.freenode.net There's also a discord channel: https://discord.gg/xhRzpPn

This is no "code" related issue.