samtap/fang-hacks #220: xiaofang hack have only sound  but not display image when i run via vlc,...

baoson202, 2017-09-03:
xiaofang hack have only sound  but not display image when i run via vlc,...
Please help me !

Chrischi-, 2017-09-03:
I think this is no "code" related issue. Try changing the resolution and or the bitrate.

For questions, suggestions or just general discussion, please join #fanghacks on irc.freenode.net There's also a discord channel: https://discord.gg/xhRzpPn

samtap, 2017-09-06:
@baoson202 Chrischi is correct, it's working for many others so likely a local problem and not a bug, which is why I'm closing it.

If you still need help you should run vlc with increased debug logging, which usually provides some hints. Post the logs somewhere (pastebin) and link to it on discord and you may find someone there who's seen it before and is able to help you.

closed 2017-09-06 by samtap