samtap/fang-hacks #221: Password protect the HTTP interface

nmaster2042, 2017-09-03:
Hi, 

I have successfully hacked 2 cams, very cool stuf you made, thanks.

Is there a way to password protect the http://DEVICE_IP/cgi-bin/status url to prevent everyone to make change other than the admin ?

samtap, 2017-09-03:
Not yet but I'm planning to switch to busybox httpd instead of boa, to support auth.

BlaY0, 2017-09-19:
You could make use of uhttpd which is used mainly by OpenWRT. You can download already precompiled binaries (Kirkwood) from OpenWRT site. The only drawback at this point is that you would need to use matching uClibc because one used by SONiX is lacking some functionality.

samtap, 2017-09-19:
httpd works find and is already available (from busybox). I would look into something else if it provides auth, ssl and cgi support. Httpd does auth and cgi but I still need stunnel for ssl.

BlaY0, 2017-09-19:
That's exactly what uhttpd supports:

https://wiki.openwrt.org/doc/howto/http.uhttpd

samtap, 2017-09-19:
I'd like to keep all scripts compatible with boa so you could also use it without sd-card (losing auth and ssl support). But I'll have a look at uhttpd and see if it builds agains snx libc and openssl (libustream may be a problem).

mvillarejo, 2017-10-03:
It'll be great to have this feature implemented. +1