samtap/fang-hacks #222: Please help! - 404 not found

kellyfw, 2017-09-08:
Hi. I'm not even a noob. I'm a nearly 50 year old lady who, up until recently, really loved her two cheap xiaofang security cameras, which worked like a dream. Then I updated my Mi home up, and they got worse. The next update was iffy, then the next rendered them unusable, and any time I try to connect them it brings my home internet to a standstill. Despite my sternly worded ratings in the app store, I now accept that Mi home is never going to do it for me. I realised that the hack which enables the RTSP thingy will be perfect, I only need to check the security cameras at home.  

OK, so I've read everything on here. sobbing gently as I had to google every acronym. I've tried:
writing the image file  to the SD card. (I didn't do the manual version, because it lost me at "create a partition"). Anyway, the camera gets the solid blue light, I put in the SD card and get the clank-clank noise. But I can't go to the device-ip/cgi-bin/status, I get a 404 not found. BTW, is that an actual external internet address, or is that just my computer talking to my camera?

Anyway, from reading up on here I diagnose that the snx thingy didn't load. But that's a problem because I need to access the device-ip/cgi-bin/status thingy to edit what's on the camera. I've tried it on both cameras, same result. They're both MAC 34s. I see from on here that sometimes people have managed to hack these, sometimes not. (I note that sometimes the fix involves holding down buttons and turning them on and off, which is about my level of expertise with all things technical).

Can anyone give me any advice? To give you some idea of my grasp of the subject, I didn't know what an IP address was. Then I knew what one was, but not how to find it. Then I found it (there's an app for that!). After the 404 message, I looked at the advice on here, someone advised to ping the device, so after googling what that meant, I managed to at least try to ping the device, which didn't work. After much research, I found that the firewall thingy was preventing the ping, because no-one wants a ping of death, right?

If anyone can give me help, it would be greatly appreciated. If I don't fix the situation, I'm going to buy two security cameras locally, at a total cost of $400. Eye watering, I know, but at least if those suckers don't work I will get my money back.

samtap, 2017-09-09:
Do you know which firmware version the cams are using? You can check in mi home. If you're using the latest version (3.2.0.30) you can try to downgrade.

kellyfw, 2017-09-10:
Thanks! It's the latest version of the firmware, so I'm now trying to downgrade it. I've found the instructions: https://github.com/samtap/fang-hacks/wiki/HowTo:-Flash-original-Xiaomi-firmware-from-sdcard-(factory-reset) I'm using the "SD card" instructions. 

I've downloaded upd_isa.camera.isc5.bin.extracted. I'm afraid I have problems with the next instruction: 
"Extract the file and place the files in the root of your SDCard on the first (FAT) partition "mmcblk0p1""

I'm a real ignoramus at this: I look at the SD card in file explorer and just see its contents. Is there a particular bit of software I should use to be about to see things like FAT partitions and mmcblk0p1? I'm using Windows. I know I can probably find this out by a bit more googling, but I've already pushed my technical knowledge to the max and I'm feeling a bit overwhelmed.

samtap, 2017-09-10:
You have to place the FIRMWARE_660R.bin file on the SDCard, right next to the snx_autorun.sh script. Windows explorer only lets you see the first partition as it can't read the other one, but that's fine. You should be able to see the bootstrap folder, snx_autorun.sh script and probably some other stuff, just drop the firmware file there. Insert the card into the camera and keep the reset button pressed while you power it on. 
[FIRMWARE_660R.zip](https://github.com/samtap/fang-hacks/files/1290171/FIRMWARE_660R.zip)

kellyfw, 2017-09-10:
OMG, that worked! That's so awesome, the mi home app is working now. Hmm, now I can decide whether to continue on with the rest of the hack, and go for the RTSP. But I'm so happy to have my cheap camera working again, thank you!

kellyfw, 2017-09-10:
I'm happy to report I was also successful in downgrading the firmware on the other camera, and then it too connected to the mi home app and worked! Hooray!

But I'm now pretty peeved with the mi home app (did they deliberately render the product unusable??), and would like to try the RTSP thing instead. When the camera has a blue light because it's connected to my wifi, If I eject the SD card (which I'd earlier put the 2.0 fang hack on to try to do the RTSP thing), it does the clank-clank noise. But I still have no luck with the device-IP/cgi-bin/status thing - the message is 404. As far as I understand what I read earlier, I need to make the connection with the device-IP/cgi-bin/status thing in order to delete old versions of the hack. Or, am I confusing things too much to try to do the hack after I did the downgrade? 

Is that device-ip/cgi-bin/status an actual external internet address, or is that just my computer talking to my camera?

Thanks again for your help. At the moment, I'm happy to have the cameras working again!

samtap, 2017-09-10:
You should boot the camera without sdcard, then insert it when the light is blue. Then when you hear the sound the status page should be available. You need the find the device-ip of the cam on your network, so it's not external. You can find it in mi home or if you get the annoying 'cannot be used outside china message' you can look it up in your router or use a different method. Then type the ip in your browser and try to access cgi-bin/status or cgi-bin/hello.cgi.

mr-varga, 2017-09-12:
Hi, i have the same issue, the 404 error page.
I'am also unable to downgrade the camera with this procedure, I turn on the camera pushing the button then the led stay fix ornage for a 40 second, after that blink blue and green and finaly blink green for the rest of the time. I see the camera ip in fing but i'am not able to apply the hack also unable to downgrade. In the mobile app I can see the camera but this is the message when i click on it: "Use only in mainland China This product is only used in mainland China."

Can you help me for the solution?

Swaktor, 2017-10-08:
DOWNGRADING to 3.0.3.56 from 3.2.0.30 And Installing Fang Hack without Error 404. !! Solution !!

I found this Tutorial Custom firmware in English Language.
https://www.youtube.com/watch?v=mpzPWYONWZA

I just put this .bin File on my FAT32 formatted 16GB MicroSD Card with only 1 Volume / Partition and downgraded it as it is shown in the Video.

After downgrade, connect your XF Camera with the MiHome App 4.0.11 or 4.0.11 to the Wifi- I prefer using .11

Then I deleted the .bin File and copied the Fang hack Master File into your root folder of the MicroSD Card.
Get it here and dowload the zip File : https://github.com/samtap/fang-hacks

Insert the MicroSD card while you XF Cam is Connected to the WiFi and then type
http://camera.ip.address.here/cgi-bin/status. Thats it !
For me there was no Error 404 and I was able to check the http://camera.ip.address.here/cgi-bin/hello.cgi