samtap/fang-hacks #223: Video recordings on the SD Card

timsayshey, 2017-09-15:
I noticed that some video files are being recorded to the SD card. Is this happening based on motion detection? Also, I am thinking about setting up a script that will SFTP the videos files then upload them to Dropbox. Just wanted to get more information about the video files first. Is there any way to control when they get recorded and will the oldest get purged if the sd card runs out of space? Thanks!

samtap, 2017-09-15:
They are created by cloud apps. There're some options in mi home to control it but I don't know the details (don't use mi home).

timsayshey, 2017-09-15:
Yeah, I'm using TinyCam/iSpy right now. 

So, it will only auto-record whenever Fang Hacks aren't running? Or when Fang Hacks are running and Cloud is enabled?

samtap, 2017-09-15:
The cloud app and rtsp server can't run at the same time. You can use either with fang hacks, just not at the same time. I'm working on a new rtsp server that supports recording. It'll be similar to cloud apps, placing video files on sdcard so they can be processed from there i.e. upload via sftp, dropbox etc.

timsayshey, 2017-09-15:
Woah, that sound amazing! Looking forward to it. Btw, thanks for all of
your work on this project.

If I come up with any improvements I'll submit a pull request.

BlaY0, 2017-09-18:
@samtap why just not use /dev/video2 for local recording since /dev/video1 is used for streaming?

peros550, 2018-02-12:
I'm also interested in the functionality of having both the RTSP and SD card recording on the same time. 

Many Thanks for your work.

peros550, 2018-02-13:
So currently is it possible to run RTSP server and have the camera recording video on the same time?

peros550, 2018-02-14:
I found my answer here: https://github.com/samtap/fang-hacks/issues/303