samtap/fang-hacks #225: http://device-ip/cgi-bin/status is not working

f3rdi, 2017-09-19:
Hi,

I am from Sofia, Bulgaria and the url http://device-ip/cgi-bin/status is not opening from here.
Can you check ? 

Thanks!

halfluck, 2017-09-20:
device-ip needs to be changed to your actual camera's ip address e.g. 192.168.0.23
so the full url in this case would be http://192.168.0.23/cgi-bin/status
you can get your camera's ip add either from the Mi Home app or your Router

Swaktor, 2017-10-08:
DOWNGRADING to 3.0.3.56 from 3.2.0.30 And Installing Fang Hack without Error 404. !! Solution !!

I found this Tutorial Custom firmware in English Language.
https://www.youtube.com/watch?v=mpzPWYONWZA

I just put this .bin File on my FAT32 formatted 16GB MicroSD Card with only 1 Volume / Partition and downgraded it as it is shown in the Video.

After downgrade, connect your XF Camera with the MiHome App 4.0.11 or 4.0.11 to the Wifi- I prefer using .11

Then I deleted the .bin File and copied the Fang hack Master File into your root folder of the MicroSD Card.
Get it here and dowload the zip File : https://github.com/samtap/fang-hacks

Insert the MicroSD card while you XF Cam is Connected to the WiFi and then type
http://camera.ip.address.here/cgi-bin/status. Thats it !
For me there was no Error 404 and I was able to check the http://camera.ip.address.here/cgi-bin/hello.cgi