samtap/fang-hacks #226: iSmartAlarms firmware - hope is useful

SpoKeys, 2017-09-21:
[4.0.4.9_iSC5P.zip](https://github.com/samtap/fang-hacks/files/1322845/4.0.4.9_iSC5P.zip)

iSmartAlarms firmware - hope is useful, I know the samtap may find this useful. This is latest firmware. I also have version 2.4.8.9_iSC5

SpoKeys, 2017-09-21:
[2.4.8.9_iSC5.zip](https://github.com/samtap/fang-hacks/files/1322850/2.4.8.9_iSC5.zip)

If needed.

SpoKeys, 2017-09-25:
Can anyone share how to extract the needed 0.elf from these files?

ykhandler, 2017-09-25:
i think it is already the firmware, you dont need to extract the 0.elf, just rename itu to FIRMWARE_660R.bin, then flash it.

piotrkpaul, 2017-09-26:
It it possible to flash iSmartAlarms firmware on Feng camera?

ykhandler, 2017-09-26:
I haven't tried it, but if anyone want to try it, i suggest to backup all the camera partitions, especially mtdblock5 that contains the device id.

grabbertdi, 2017-10-14:
Woohoo, thanks very much I am on it, hopefully can make it work in a few days :)

piotrkpaul, 2017-10-16:
@grabbertdi looking forward for your update :)

robaq, 2017-10-19:
@grabbertdi have any news? can you update us on your tests? I hope this can be the ultimate solution to using the cam with MI Home or the SmartAlarm app. Thank you very much

grabbertdi, 2017-10-20:
I am working on it, maybe a manual firmware flash needed, if I am up and runnign I'll let you know.

nimdy2017, 2017-10-23:
I have a device that is possibly bricked. Would it be worth trying to see if flashing the ismartalarm firmware will make it work? I haven't got much to lose really as it isn't working currently.

robaq, 2017-10-23:
@grabbertdi Many Thanks. Good work !

piotrkpaul, 2017-12-11:
any progess guys?