samtap/fang-hacks #227: 21-ir-control	NOK

TheMym, 2017-09-24:
Hello,
I updated the fang-ir-control.sh to keep the cam cool but now when I check the scripts it says the 21-ir-control is not ok. Anybody else have this problem ? The night vision is still working great but I just want to know if this is normal.
Thanks !

kslmgd, 2017-09-24:
must use reboot from fang hacks webpage after making the change.

toomuchio, 2017-09-24:
Make sure it's not dos formating as well.
You can use vi on the camera if you ssh in to edit the file, or sshfs to another Linux machine.

TheMym, 2017-09-24:
@kslmgd I did reboot it as well but no go.
@toomuchio I just edited the file with notepad++ and overwrote the old one before rebooting.

DarkStorm99, 2017-09-24:
I had the same issue. I made a copy of ir control first. Ssh into the unit. Copied the new ir script to clipboard, opened original ir control with vi, and pasted content. This appeared to resolve the issue. I suspect maybe a permissions problem on the new file causes issues

ykhandler, 2017-09-30:
If you want to transfer an .sh file dont use ftp, sometimes it change the format of the file. It is the return character problem. If you still want to use ftp, rename it without the .sh extention then transfer it, then rename it back to .sh if it is already copied.

TheMym, 2017-10-07:
Thanks will try the sh trick !

PrawnMan, 2017-11-14:
I cant seem to find the IR script: 

https://raw.githubusercontent.com/samtap/fang-hacks/master/data/usr/bin/fang-ir-control.sh

The link seems broken? 

Anybody have a link to the updated script? Is it in the current image?

jivonl, 2018-01-02:
i had to change the " permissions "  on the file.

Login with ssh on the device:
`cd /media/mmcblk0p2/data/usr/bin`
`ls -l | grep fang`
**`-rw-r--r--    1 root     root          1433 Dec 19 09:31 fang-ir-control.sh`**
`-rwxr-xr-x    1 root     root           818 Jan 25  2017 fang-ir-control.shbackup`

this bold is the new file with wrong permissions (check with older file below)
in this case the resolution was to use the next command
`chmod +x fang-ir-control.sh`
check with:
`/media/mmcblk0p2/data/usr/bin # ls -l | grep fang`
`-rwxr-xr-x    1 root     root          1433 Dec 19 09:31 fang-ir-control.sh`
`-rwxr-xr-x    1 root     root           818 Jan 25  2017 fang-ir-control.shbackup`

now the script is " visible " and status OK

a-triant, 2018-09-01:
Hello!

I have same issue, I overwritten the fang-ir-control.sh(/media/mmcblk0p2/data/usr/bin) with the edited one, using a FTP client (FileZilla), also through SSH as said above I have set 755 permissions to the updated script. However the IR does not working after that, and the 21-ir-control shows NOK. Probably already reseted the camera and reformatted the SD card for over 15 times, tried disabling enabling the script, rebooting the camera many times, nothing helps.

I don't understand what i'm doing wrong, any ideas?

Thanks.