samtap/fang-hacks #228: 我删除了/etc/fang_hacks.sh、/etc/fang_hacks.cfg、/etc/fang_hacks.cfg.bak，重启后一直黄灯怎么办?

LiuJiangshan, 2017-09-24:


closed 2017-09-24 by samtap

LiuJiangshan, 2017-09-24:
??

ykhandler, 2017-09-25:
Just flash a genuine firmware or a custom firmware. Look at this video to do it:
https://youtu.be/mpzPWYONWZA

SpoKeys, 2017-09-25:
Can you share the firmware you have used, or some details about it?

ykhandler, 2017-09-25:
Of course. I will share it on my youtube. But currently i'm designing the easiest way so every body will understand it.

SpoKeys, 2017-09-25:
Please try any way you can. Also I see the clip is restricted outside the app(not available on browser) so comments are hard to get. 

Probably there are more people like me that would like to know how to get 0.elf from a .img file that is the default file format for firmware upgrade. I am not really an IDA user but I`ll do whatever it takes.

samtap, 2017-09-25:
I'm removing this thread. I've no idea what's going on from chinese title to someone trying to lure people to his youtube channel. Please be clear when you open a new issue, if you just have a question or need support drop by gitter chat.

ykhandler, 2017-09-25:
Peace Samtap. Simple answer: Flash the Firmware to recover your camera LiuJiangshan...