samtap/fang-hacks #229: Update procedure

james-fry, 2017-09-26:
I have check the readme.md and wiki/faq etc but I can't see what is the procedure to update.
I am using the 0.2.0 image, which is ~9 months old.
I can see the latest commit is ~16 days back, so the image is old!

What is the procedure to upgrade the camera to latest hack?
I am using windows, so mounting the sdcard ext2 partition will be unfun. 
I can ssh/scp to the device and can see the sdcard contents in /media/mmcblk0p2/data
Should I simply copy files onto this location and overwrite?

djsuszi, 2017-09-26:
maybe someone cen build new image ?

james-fry, 2017-09-27:
They might, and that would be great... but I'm not holding my breath :)
I'm sure there must be a well worn procedure to update the files on the device or SD card.
I'm just being careful in case there is a mistake and I brick the cam.

thundergreen, 2017-09-28:
I was also wondering how to update.. can the dev please give a short feedback?

samtap, 2017-09-28:
There're not many relevant changes pushed to the repo, but feel free to manually update stuff. I recommend waiting for a new image release. I'm contemplating an auto-update mechanism (so you could pull changes from github instead of writing a new image every time something is fixed)

thundergreen, 2017-09-28:
Apparently here's something new to the ir script. I'm asking because my cam runs hot and stops steaming after a while. Do u think by disabling all other services the cam could stay cooler? I only need the rtsp stuff

james-fry, 2017-09-28:
This is the exact same reason I asked!
I am finding that the rtsp server is a bit flakey, and somewhere else it was attributed to heat.

samtap, 2017-09-28:
Some cams have the heat issue, others don't. There's an updated script which resolves it, but it's not perfect either.
You can download the updated file: https://raw.githubusercontent.com/samtap/fang-hacks/0d2913f11b6bcf471f97e7bb29e6d4b5777c1dbf/usr/bin/fang-ir-control.sh
You can copy it to the device (/media/mmcblk0p2/data/usr/bin) using ftp or sftp but make sure the file-attributes are correct (often ftp clients drop the executable attribute). You can fix this by logging in over telnet/ssh and running 'chmod 755 /media/mmcblk0p2/data/usr/bin/fang-ir-control.sh