samtap/fang-hacks #230: Static orange LED after applying hacks

drschnalli, 2017-09-28:
I just wrote the newest release via .img file to a microsd card and applied the hacks. I disabled Cloud functionality and clicked the button to extend the partition. Its a model with 28: mac address.

After reboot the device was not reachable anymore and the status led is static orange. 

I tried to follow the instructions of reflashing the original firmware but nothing happens. I left it for about one our with the sd card (powered on while i was holding the config button for 10 secs). 

Do i now have to open the enclosure to get access to the internal serial port to manually invoke firmware flashing?

aderusha, 2017-10-05:
I have run into this exact sequence of events (apart from having a 34:CE:00:xx:xx:xx MAC).  Come to find out disabling cloud apps isn't a great idea.  I've soldered on a 232 interface and I'm able to login with root/ismart12 but I have no idea how to re-enable networking.

brodo97, 2017-10-19:
@aderusha Did you solve the problem? I did it today, it's not difficult.

aderusha, 2017-10-19:
No I never did!  If you have any guidance I sure could use the help.

brodo97, 2017-10-19:
@aderusha I simply edited with vi file "/etc/fang-hacks.cfg" changing from DISABLE_CLOUD=1 to DISABLE_CLOUD=0. Also look at NETWORK_MODE. I set NETWORK_MODE to 0 and now it connect to my wireless network

aderusha, 2017-10-19:
I'll give that a shot when I get home tonight - thanks man!