samtap/fang-hacks #232: Cannot downgrade at all

superdic, 2017-09-28:
Dears,

I tried everything but I cannot downgrade at all my xiaofang camera to a suitable firmware. My cam is locked to be used outside China.

I think I tried everything but nothing at all. What I simply did is:

1) Creating an SD card using the image 2.0
2) copying the 0.elf renaming it to FIRMWARE_660R.bin
3) Inserting the SD Card and powering on the camera keeping pressed the reset button.

So nothing at all.. always rebooting with latest version and not the old one....  I'm using a 32gb SD.

Could someone explain to me where I'm failing? I'm really descouraged !

Thanks a lot !

45Wheelgun, 2017-09-28:
I just sucessfully did downgraded this morning.  I did the following steps and I notice that at one point it appears I deviated from what you list above and that might make the difference.

1. Created an SD using image 2.0
2. Downloaded the downgrade ROM and unpacked it.
3. I copied the entire package (not just 0.elf) to the SD card root. 
4. Renamed 0.elf
5. With Power 0ff - I inserted SD
6. Using pin provided with camera I held in the setup switch while connecting the power.  I held that pin longer than I thought I should have to.  (Slow count to 20) 
7. Let go of the pin, walked away, got coffee and purposely didn't go back for 10 minutes.

I noticed my #3 differs from your #2 which seems to indicate you only copied 0.elf to the drive.

My #6 and #7 were both excessive in the amount of time I waited.  I have found over the years that I personally can be impatient and mess things up because I think either nothing or that it isn't happening fast enough.  All I know is I didn't do it right the first time and when I followed the steps above it worked.

superdic, 2017-09-28:
Thanks for your feedback!

Question, did you remove the files of the image or you left them in the partition?

45Wheelgun, 2017-09-28:
I made the image and left everything in place and added the downgrade files.

ykhandler, 2017-09-29:
For downgrading, there is no need to create the sdcard using image file. All you need to do is put the FIRMWARE_660R.bin into sdcard (use not above than 32gb). Make sure it is FAT32 formatted if you have been changed it before.

Turn off camera (unplug from power source)
Put the sdcard on the camera, hold the reset button, turn on the camera, count to 10, release the reset button. 
The camera will yellow still for about 1-3 minutes. Just wait for it. If success, it will blink yellow, and if you press the reset button once, it will make a sound.

45Wheelgun, 2017-09-29:
Totally agree, I just used the image so I knew it was partitioned and formatted correctly.

superdic, 2017-09-29:
I'm quite sure I've used this approach as I tried everything.

Anyway today I will try again following step by step your suggestion. I'll let you know the results.

Talk u later

superdic, 2017-09-30:
So, guys, I tried and seems that something happened this time but..

1) I'm not able anymore to pair the camera with Mi HOme (always in time out)
2) I cannot see anymore the Camera SSID....

any thought?

ykhandler, 2017-09-30:
Try to remove space in WIFI SSID and password if you have it. Or maybe try to use other wifi source. Have the camera turn blue when you try to pair it?

superdic, 2017-09-30:
yes it turned blue and yellow blinking... but looking at the router it seems that it caught the IP Address...

ykhandler, 2017-09-30:
Do wifi ssid or password got space character? Try to change it if it has it.

superdic, 2017-09-30:
no Space chars.

there is something strange.....

ykhandler, 2017-09-30:
Then, try to use other wifi source ( other provider ). I hope the camera id is not gone...

superdic, 2017-09-30:
wifi is working properly. Now the camera has the IP address from my router. It's pingable too.

But I cannot pair it with MiHome....

ykhandler, 2017-09-30:
I know that. It even can access internet, but it cannot reach the xiaomi server (if the camera id is not gone). 
If you got 2 cellphone, try to use 1 of it as hotspot.

Swaktor, 2017-10-08:
DOWNGRADING to 3.0.3.56 from 3.2.0.30 And Installing Fang Hack without Error 404. !! Solution !!

I found this Tutorial Custom firmware in English Language.
https://www.youtube.com/watch?v=mpzPWYONWZA

I just put this .bin File on my FAT32 formatted 16GB MicroSD Card with only 1 Volume / Partition and downgraded it as it is shown in the Video.

After downgrade, connect your XF Camera with the MiHome App 4.0.11 or 4.0.11 to the Wifi- I prefer using .11

Then I deleted the .bin File and copied the Fang hack Master File into your root folder of the MicroSD Card.
Get it here and dowload the zip File : https://github.com/samtap/fang-hacks

Insert the MicroSD card while you XF Cam is Connected to the WiFi and then type
http://camera.ip.address.here/cgi-bin/status. Thats it !
For me there was no Error 404 and I was able to check the http://camera.ip.address.here/cgi-bin/hello.cgi

nimdy2017, 2017-10-13:
@Swaktor - The audio changes to English when I use this firmware, so something is happening. Whether I use this or the official .bin file I cannot get the device to recognise my WiFi. Any ideas?

Swaktor, 2017-10-13:
@nimdy2017 did u Check  disable Cloud in the Fang Hacks accidently . This Can Brick your camera.

nimdy2017, 2017-10-13:
@Swaktor  I've got two cameras. A 34 with button worked fine, but the 28 with the pin hole is giving me problems. I followed the guide and got to the hacks webpage. I pressed the button to expand the SD card partition and restarted the device. Since then I haven't been able to connect it to the WiFi at all, either with the app or with the hacks. I get an alternating blue and orange light. The Mi Home app recognizes the devices but isn't able to configure it. Does this sound like a bricked device?

Swaktor, 2017-10-13:
@nimdy2017 are you able to Ping the camera? Is the app saying that it is online? Is the camera blinking blue ?

nimdy2017, 2017-10-13:
@Swaktor No to both. I use an app called fing on my phone to see devices connected, but it doesn't show up. The light blinks orange and blue.

Swaktor, 2017-10-18:
@nimdy2017 it seems the camera is bricked.  That's what happened to my first camera.

ykhandler, 2017-10-18:
Make a video and upload to youtube...

nimdy2017, 2017-10-18:
@Swaktor No way at all to unbrick?

nimdy2017, 2017-10-24:
I've got it back. It will only connect to my mobile hotspot and not my home WiFi now. Trying to work out why.

Swaktor, 2017-10-27:
Unbrixk with a rs232 to usb ja possible . Theres YouTube Video with unbrick xao Fang camera

YENANCHEN1021, 2017-11-09:
I have problems while downgrading from 3.1.0.40 to 3.0.3.56.
My firmware might be a bit different than what you have because I purchased Xiaofang in Taiwan. 
Though I tried all the methods including the one mentioned above, nothing really happened after pressing the reset button for 10-15 sec. 
It just turn the light from yellow to blinking yellow(in 2 sec).
When I connect the camera from the app, the version of it never changed. 
I am so desperate... Please help...

ykhandler, 2017-11-09:
Create a dump of all partitions for us to examine. Use UART to do that.

YENANCHEN1021, 2017-11-09:
Hi ykhandler
Thank you for replying. I will try to get that device(UART) and get the data out for you guys to check.

ykhandler, 2017-11-09:
Ok, for now read this:
1. Use original adaptor and cable (insufficient power makes camera not load the sdcard).
2. Use sdcard 4GB -32GB is prefered (because their default file system is FAT32). 
3. Never forget to safe eject the sdcard from your PC before unplug it, if you dont know how to do this, turn off your PC before unplug it. (This is to prevent data corruption). You need to scan disk your sdcard if you have unplug the sdcard before do the safe eject procedure,
4. The filename must be FIRMWARE_660R.bin (it is case sensitive, so make sure the upper case and lower case is in the right place).  And if you download it multiple times, you must see there is '(1)' in the file name, so you need to rename it (ex: FIRMWARE_660R(1).bin).
5. If the camera has a corrupt partitions, you need to use original firmware that include uboot and kernel partitions, before applying my firmware.
6. Dont forget to press and hold reset button, before turn on the camera, then count 10 seconds after you turn on the camera before you can release the reset button.﻿

YENANCHEN1021, 2017-11-09:
Thank you for the detailed tutorial.
I did follow all the steps in it but still doesn't work.
I guess the version on my camera cannot read FIRMWARE_660R.bin file so it just turned on like a normal start.
About the UART device, do you mean UART-TO-USB device?

ykhandler, 2017-11-09:
Yes it is. Also known as usb to TTL.

alcestneige, 2017-11-10:
@Swaktor Thanks man for the small tutorial. I have the lates camera with 34 mac address and china only. Hacked after 3 minutes :D I'm in europe and the camera is working great. First downgrade, then install the MiHome App 4.0.11, connect to wifi and finally create the sd card with the hack and you are good to go. Everything is working flawlessly. Only one question. The cloud is disabled by default? or do I need to disable it somehow? As I do not want to use it at all. I would like to connect it with NAS if that's somehow possible. Thanks again! I was afraid that the china only version will not work but it does.

superdic, 2017-11-11:
Guys,

I tried in any possible way but nothing... I'm not able to downgrade. I also tried what @Swaktor suggested but nothing....

I tried with several SDs. what happen now is that the light stay on just for 10secs after I leave the button and it starts blinking quite immediately... I'm really frustrating...

YENANCHEN1021, 2017-11-15:
@superdic 
It seems like you encountered the same problem.
My last camera was from Taiwan so I try to get one from China and it just arrived. 
When I follow the tutorial above, it still doesn't work. 
The light stayed when the reset button was pressed. And when released the button, the light started to blink around 2 seconds later. 
I guess they must find some ways to block us from loading our own firmware in.

ykhandler, 2017-11-15:
I believe there only 3 causes:
1. File corrupt.
2. Sdcard corrupt.
3. Camera power supply insufficient.

YENANCHEN1021, 2017-11-15:
@ykhandler 
I am not so sure. 
I tried two brand new SDcards and using the attached USB power supplier but still doesn't work.
About the file problem, could you please provide a link for me to try?
Thank you.

ykhandler, 2017-11-16:
https://www.dropbox.com/s/kzyxjdpxmyk3fjs/FIRMWARE_660R.bin?dl=0

then check for integrity:
CRC32             Size  Name
-------- -------------  ------------
7EA476A6       8212032  FIRMWARE_660R.bin
-------- -------------  ------------

renantins, 2017-11-19:
I tried this tutorial but doens't work https://github.com/samtap/fang-hacks/wiki/HowTo:-Flash-original-Xiaomi-firmware-from-sdcard-(factory-reset) . I have a 16GB SD card formatted in FAT32. What cat it be?

YENANCHEN1021, 2017-12-08:
Hi @ykhandler 
I got the USB-UART device and teared my camera just yet. 
Could you let me know which software you use to communicate with your camera please?
If there is a tutorial for me to follow that will be great!
Thank you.

ykhandler, 2017-12-09:
Use putty, and find your usb uart part at device manager.
After all connected, turn on putty, setup the port number, klik ok on putty. Turn on the camera, if it is all ok, you will see the text in the console.

tdtgit, 2018-01-02:
I'm from 3.3.3.25, MAC 34. Tried both @ykhandler and from [Wiki](https://github.com/samtap/fang-hacks/wiki/HowTo:-Flash-original-Xiaomi-firmware-from-sdcard-(factory-reset)), maybe firmware problems :(