samtap/fang-hacks #233: RSTP server by demand only

350d, 2017-10-03:
Is there any options to start RSTP server and IR LEDS by demand only (Only if someone connected to the camera)? I don't need to record any videos most of the time, so, no need to heat up the camera. Will be awesome to have this option here.

samtap, 2017-10-03:
Nope. You can stop and start rtsp server using the web-interface.

350d, 2017-10-03:
Yes, i known about manual options, but i'm Homekit/Homebridge (like many others) user and looking for any options to have RSTP stream only if i check my camera time to time. This feature request sounds legit, isn't it?

samtap, 2017-10-03:
No it sounds like the wrong solution for a different problem.

masterchop, 2018-06-04:
this is the command executed maybe you want to try around to see which flag disable recording

snx_rtsp_server -W 1920 -H 1080 -Q 10 -b 4096 -a

report back if you have success.