samtap/fang-hacks #234: Support DaFang?

jejer, 2017-10-10:
Hi,

Will you support DaFang?
I want to use RTSP directly without MiHome.

Thanks!

samtap, 2017-10-10:
I'm looking into it. I have one, and succeeded in getting serial console access. I've also intercepted a firmware update, but it's only a partial update not a complete firmware image.

Unfortunately it doesn't share any of the hardware components with XiaoFang. It is MIPS based instead of ARM, and has a different SoC. There don't appear to be many security provisions though (i.e. no signing like on Mijia1080p). It runs a telnetd on boot, which get killed immediately when Xiaomi software is started, so I don't have a simple 'jailbreak' method yet that doesn't require getting serial console access.

Any help or info is appreciated!

ykhandler, 2017-10-10:
Try creating script in /etc/init.d
Ex: S07service
Contains: 
#!/bin/sh

busybox telnetd &

samtap, 2017-10-10:
Sure, how do I create that without having serial console? ;-)

ykhandler, 2017-10-10:
"I have one, and succeeded in getting serial console access", i thought you already have serial console access. Please share the firmware update file, maybe it has a clue.

samtap, 2017-10-10:
I have serial console access but you can't expect regular users to take it apart and solder wires to the pcb.

We need to find something like the snx_autorun.sh trick in the older XiaoFang firmware. But the DaFang doesn't appear to use any scripts for mounting, same as on latest XiaoFang firmware they've integrated mounting into their executable instead of handling it using regular hotplug kernel facilities.

ykhandler, 2017-10-10:
I see... may be someday i'll buy this camera and take a look of it...

robinson4, 2017-10-13:
I just got one of these. I'm keen to get basic RTSP or even still image capabilities as well as GET requests to control PTZ functions.

I'm happy to help out with device testing etc. I have FTDI gear also.

EliasKotlyar, 2017-10-15:
@samtap: Could you please provide a picture of the Pins on the Board? Will get my DaFang soon and want to try to investigate some details about the hardware.

jimmyktp, 2017-10-17:
One of the reviews out there. https://www.youtube.com/watch?v=nDdnphZF6Cw

Seems like some function is not working. Currently, the latest firmware is 5.5.1.177, anyone tried that? I dare not to update the original firmware as I'm worried it might get shittier.

EliasKotlyar, 2017-10-20:
Hello everyone,

Today i got mine. I havent managed to connect it yet, but have made a teardown. I collected all the informations in a new Repository.
https://github.com/EliasKotlyar/Xiaomi-Dafang-Teardown

Feel free to contribute any information about the dafang.

Greatings,
Elias

samtap, 2017-10-20:
Did you get a serial console? The pads are visible here https://github.com/EliasKotlyar/Xiaomi-Dafang-Teardown/blob/master/teardown/mainboard3.jpg on the side of the board you can easily add a header for gnd, rx, tx and route some wires through the bottom of the cam.

EliasKotlyar, 2017-10-20:
Hello samtap,

Havent tried the serial connection yet. I need to get a FTDI-Adapter to try it out, and i will get it hopefully soon.

I just want to clarify about the serial connection:

Is the order of the Pins correct?
https://github.com/EliasKotlyar/Xiaomi-Dafang-Teardown/blob/master/teardown/headers_sample.jpg

Could you please explain which bitrate and voltage you have used? I assume it is:
115200 Bits and 5V. 

Also i found out that there are some ports open which can be used to control this device. Will try MitM if i have time. Maybe we can control it using simple commands.

Greatings
Elias

samtap, 2017-10-21:
Yes 115200, the level is 3.3V, I don't remember order of tx and rx pins but you can just switch them when it doesn't work ;)

beikeland, 2017-10-21:
Would it be feasible to add a bluethooth serial module and find 3.3V on the board to avoid having to run wires. Anyone identified a 3.3V pad?

I'll gladly add serial headers - FTDI / Bluetooth to help test stuff when I get mine.

EliasKotlyar, 2017-10-21:
Hello everyone,

I managed to get a FTDI with 3.3V and got a connection to the uboot. I have documented my way in my Repository.

Here is a bootlog of the device:
https://github.com/EliasKotlyar/Xiaomi-Dafang-Teardown/blob/master/mod_serial/booting.txt

Will try to dump the firmware soon/get a Telnet running. Any tips on that @samtap ?

@beikeland : It should work, there are a lot of 3.3V pads inside. Have a look on my solution - I just drilled a hole and routed some wires inside. 

Greatings
Elias

Kadalia, 2017-10-21:
Hello,
I've just received a couple of Dafang ... unable to use then !
Read some forum comments, it seems that these cams are dedicated to china market :-(
definitively need a hack :-)
Go Elias Go !
I dont know how to help you .... so if I can do something for you, then just ask !
Thank you,
Philippe

samtap, 2017-10-21:
If you disconnect flatcable to the sensor pcb, iCamera doesn't start since the sensor chip is missing, and you can see telnetd is running. It is killed by iCamera when it starts.

nateinaus, 2017-10-22:
Hi, new to github so im not sure if sharing a file this way will work. I just received one of these and it turned up dead, I managed to track down the firmware from a guy on youtube. Not sure if its the full thing or not but it seemed to work for him. My camera doesn't even light up at all. The board is getting power and it warms up but nothing else. Hope this helps.
[demo_5.5.1.194.zip](https://github.com/samtap/fang-hacks/files/1404551/demo_5.5.1.194.zip)

EliasKotlyar, 2017-10-22:
Hello everyone, 

I have several interesting news:

1. I have dumped the firmware and uploaded it here for further investigations.
2. There is some code in the bootloader (u-boot) which can make an whole update from the microsd. Unfortunately , i was not able to find out how to trigger it. Maybe someone can have a look? It starts with "jiabo":
jiabo_do_auto_update!!!!!!!!!!!!!!!!!!!!!!!!
jiabo_au_check_cksum_valid!!!!!!!!!!!!!!!!!!!!!!!!
jiabo_update_to_flash!!!!!!!!!!!!!!!!!!!!!!!!
jiabo_au_do_update!!!!!!!!!!!!!!!!!!!!!!!!
jiabo_idx=%d
jiabo_start=%x,jiabo_len=%x

I tried the binary from @nateinaus , but this code dont see it. 
@nateinaus : Can you please tell us the video of this guy on youtube?
@all: Could someone please ask Xiaomi if they can pass us the source of the u-boot? Its GPL, so they have to.  I know there is little chance that they will do it, but we still can try.

3. I have managed to crosscompile a dropbear server, but i cannot login. Its always says that it cannot find my user:
TRACE  (176) 29.525551: enter checkusername
TRACE  (176) 29.525586: leave checkusername: user 'root' doesn't exist

I dont see any error. Maybe someone can check?

Here is my build of the Dropbear:
https://github.com/EliasKotlyar/Dropbear-Dafang

Greating 
Elias

Kadalia, 2017-10-22:
Hello,
I have "chat" with Xiaomi US team, they told me that Dafang was not yet available in US ... So they can't help me, and suggest me to mail to service.global@xiaomi.com.
Done ... wait for the source code of u-boot :-)
Very little chance .... but who knows !
Philippe

nateinaus, 2017-10-23:
Yeah I thought it might just be an update to the firmware or something. I emailed xiaomi directly asking for the uboot but not sure if I'll have any luck. The guy on YouTube was a Spanish guy.

https://youtu.be/8ouH4_Pq4Ng

ykhandler, 2017-10-23:
Nice,

Sniper-Wolf, 2017-10-23:
Hi there! 

Just offering my modest help : 
-> I can't help a lot with serial connections, but once we can get an easy telnet/ssh access, I can probably help with some scripts, dropping chinese server ip connections, replacing chinese voice files by english ones, and so on.

I messed around with the xiaomi mini router (i've since thrown it away, too much bugs). 

Here's a backup of the demo firmware from the video posted by nateinaus, just in case : 
https://drive.google.com/open?id=0B4WQ7BDk2rtZYy1Hb1dfaWE0TzA

beikeland, 2017-10-23:
Just received my first camera; took some step-by-step pictures while opening it https://photos.app.goo.gl/Yjwh9TcIZiwLfXwL2

Is U2 / 25Q127CSIG  on the bottm an spi flash containing the firmware? Google didn't turn up much info on the part name, but could try a SOIC clip or desolder it and stick it in my CH341A and see if its able to pull some data.

EliasKotlyar, 2017-10-23:
Hello everyone,

@Sniper-Wolf : Could you please have a look at my attemp to compile dropbear? 
https://github.com/EliasKotlyar/Dropbear-Dafang
I had the problem that could not login sucessfully. After that i found some patches(https://github.com/zcutlip/dropbear-hacks) and applied them. Now i can get past the login, but its segfaulting, and i cannot get reason for this error.

@beikeland I suppose it is the flash, but i am not sure. But you dont need to insert it into your programmer, because i already dumped the firmware here:
https://github.com/EliasKotlyar/Xiaomi-Dafang-Teardown/tree/master/firmware/5.5.1.177

Here are some instructions for dumping it with serial connection:
https://github.com/EliasKotlyar/Xiaomi-Dafang-Teardown/blob/master/mod_getroot/getroot.md
https://github.com/EliasKotlyar/Xiaomi-Dafang-Teardown/blob/master/mod_getroot/firmware-dump.md

nateinaus, 2017-10-23:
Is there some way to reflash the whole firmware onto this camera using the sd card slot? I tried just copying all the files on and holding the setup button as I plugged it in, but nothing happened. Cheers

EliasKotlyar, 2017-10-23:
Hello @nateinaus,

Unfortunately this is not possible at the moment. I have found some code in the bootloader(u-boot) which can do this, but i dont know how to make it work. If we would find a way to achieve this, it would be a huge step to a user-friendly flashing method. For now i can flash it only using serial-connection.

Greatings,
Elias

beikeland, 2017-10-23:
@Sniper-Wolf if you'd be interested in remote access to the serial console I'd be happy to hook you with ssh access to a host with access to the serial console of the camera if it'll help us make progress. My email should be visible on my github profile for a while.

EliasKotlyar, 2017-10-25:
I finally got this "firmware" from @nateinaus flashed on my Dafang. Many thanks @nateinaus!!

Here is a howto and my log:
https://github.com/EliasKotlyar/Xiaomi-Dafang-Teardown/blob/master/firmware/howto_firmwareflash.md

Warning: Dont try to do this unless you have a Serial Connection to rollback to the old one. I havent tested this firmware yet, i only know that its working for now.

I suppose that its not hard to modify it, because its a simple "uImage" Format.

bilseto, 2017-10-26:
@EliasKotlyar for the demo firmware, have you tried the standard xiaofang root password?

Sniper-Wolf, 2017-10-26:
Nice work Elias!

It's indeed an uImage file, i tried my luck on my Ubuntu VPS :
## First, you need to install u-boot-ools : 
_:~/dafang# apt install u-boot-tools_

Then download the firmware in a folder, and go into that folder.  (be sure not to rename it)
_:~/dafang# ls
demo_5.5.1.194.bin_

Then use this command : 
_:~/dafang# mkimage -l demo_5.5.1.194.bin
Image Name:   jz_fw
Created:      Sat Oct 14 05:03:40 2017
Image Type:   MIPS Linux Firmware (uncompressed)
Data Size:    11075584 Bytes = 10816.00 kB = 10.56 MB
Load Address: 00000000
Entry Point:  00000000
_

So we now know it's a linux firmware, from the 14 of October. It seems to be a single file image, as I do not have a "Contents" attribute. That's a kernel image, not a ramdisk image. 

We should be able to modify it but i haven't figured out how yet (I'm probably not the best for this task though).

Here's a link to help, but it's a bit different from our file : 
http://www.isysop.com/unpacking-and-repacking-u-boot-uimage-files/

samtap, 2017-10-26:
The root password is the same as on XiaoFang: ismart12

ykhandler, 2017-10-26:
@Sniper-Wolf, as i see the file. It contains kernel, root partition and jffs2 partition. So it is not only kernel image. And no kernel image filesize 3/4 portion compare with the flash space.

Sniper-Wolf, 2017-10-26:
@ykhandler : Great, what do you use to see those partition? Anyway to start modifying file in them, for example to try and keep the telnet alive?

ykhandler, 2017-10-26:
Binwalk is the best i think... I got my dafang delivery delayed because custom checking. It should be arrived in 2-3 day from now. I'll see what i can do. Can't promise anything.

EliasKotlyar, 2017-10-26:
I suppose that its in the same format as the normal firmware-dump:
https://github.com/EliasKotlyar/Xiaomi-Dafang-Teardown/blob/master/mod_getroot/firmware-dump.md

kaputtzich, 2017-10-26:
One thing that's relatively easy is to extract two squashfs files out of the bin file und unpack them
`dd if=demo_5.5.1.194.bin bs=1 skip=2097216 count=3332360 of=file_1.squashfs`
`dd if=demo_5.5.1.194.bin bs=1 skip=5570624 count=564150 of=file_2.squashfs`
`unsquashfs file_1.squashfs`
`unsquashfs file_2.squashfs`
Now we have the filesystem...

kaputtzich, 2017-10-26:
...or just use `binwalk -Me demo_5.5.1.194.bin` if too lazy :)

EliasKotlyar, 2017-10-26:
Hello everyone,

I wroten a script to extract the firmware into parts:
https://github.com/EliasKotlyar/Xiaomi-Dafang-Teardown/blob/master/firmware/howto_unpackfirmware.md

Funfact:
The md5 of the kernel and the rootfs of my dumped firmware(5.5.1.177) matches with the binarys from the new "demo_5.5.1.194.bin". The appfs and the drivers different.

Greatings
Elias

manjerico, 2017-10-27:
Hi all,

I have a couple of these Dafang Cameras and, although I can get them to work, sometimes I can't connect to them via 3G/4G.

Needless to say, that I would 1000000% prefer to connect directly to the cameras rather than via Xiaomi apps.

Unfortunately, I have no knowledge about programing/electronics to help, but I just wanted to send out my thank you and appreciation to those of you who are working on this!

If I can help in any other way, please let me know! I'll be hovering above this topic like a hawk. :p

Thanks!

beikeland, 2017-10-27:
I was under the impression you had to have xiaomi router to access the cameras through the Mi app with the stock firmware?
(or is it just a matter of supporting uPnP?)

manjerico, 2017-10-27:
I have a regular ISP-provided router. In my case, I can access the video feed and control the camera pan and tilt via Wi-Fi, when in the same network, every time (if you see in the MiHome App it calls it a Network Connection)

The problem is that when I try to connect via Mobile Data or via another network, sometimes it doesn't work. The Mi Home App just keeps on saying "Connecting (1/3)" and times out I guess. At the moment it's working fine, but later it might not. Some say that this is due to the fact this Remote function is not working well outside of China.

Nevertheless, and as I said, if a hack could be created, that would be golden! Accessing the camera directly instead of via the Mi Home app would be great.

If I can answer anything else, let me know. Thanks!

ykhandler, 2017-10-27:
Ok, i think i successfully create the firmware. Will post it as soon as possible.

EliasKotlyar, 2017-10-27:
Hey @ykhandler 

I have compiled the old firmware(5.5.1.177) . I need to try to flash it, but basically thats it:
https://github.com/EliasKotlyar/Xiaomi-Dafang-Teardown/blob/master/firmware/packer.py
https://github.com/EliasKotlyar/Xiaomi-Dafang-Teardown/blob/master/firmware/demo_5.5.1.177.bin

Greatings
Elias

ykhandler, 2017-10-27:
@Elias, ok, You have to backup all partition before you tried it. I dont have the SDK, so i cannot compile the firmware, i just do modifying the file.

ykhandler, 2017-10-27:
Btw, @EliasKotlyar have you manage to start the telnet?

ykhandler, 2017-10-28:
i've update the FIMWARE with telnet enabled. I think that's it from me, i leave the project to you guys. Hope fully getting an RTSP, control the movement using telnet or else.

EliasKotlyar, 2017-10-28:
Hello ykhandler,

Thanks for your firmware. As far as i have seen, you have modified the rootfs and integrated different soundfiles and a RC-Script for the Telnet:

```
#!/bin/sh

while [ 1 = 1 ]
do
sleep 1
result=`ifconfig wlan0 | grep UP`
if [ "$result" = "" ]
then
	cat /proc/uptime > /tmp/Handlerlog.txt
	echo "Running telnetd " >> /tmp/Handlerlog.txt
        echo "HANDLER RUNNING THE TELNETD now...."	
	/bin/busybox telnetD & 
	echo "telnetd should be Running, now killing check " >> /tmp/Handlerlog.txt
	break
	fi

done
```

ykhandler, 2017-10-28:
@EliasKotlyar, yes i did, also i modify the string telnetd to telnetD in the busybox so that the iCamera cannot kill it.

Best regard

Handler

kvanlier, 2017-10-29:
Hi @ykhandler,

Thanks for doing this.

Regards,

Krs

manjerico, 2017-10-30:
Did you guys find something in the firmware that explain why sometimes you can't connect to the camera remotely and receive the video stream?

Thank you all once again!

darioxxx89, 2017-10-30:
Hi guys, i'm very interested too for RTSP on this xiaomi dafang. This camera it's very cheap and very cool! The rotation works well, and if it is possibile to make rtsp fully working on this camera, i think it would be awesome. Any news about custom fw? I'm thinking to contact xiaomi and ask if they can add rtsp support maybe in future fw...it could be good idea?

manjerico, 2017-10-31:
Any progress with this one? :)

samtap, 2017-10-31:
Not until we get the SDK... And Ingenic have been ignoring my emails for weeks so I don't think we'll get it from them.

EliasKotlyar, 2017-11-01:
Okey, i found something interessting that might help us:
On the chinese site, you actually can buy a development board with the T20 CPU:
http://www.ingenic.cn/?product/id/24/lm/3.html
https://item.taobao.com/item.htm?spm=a312a.7700824.w4004-14666727108.8.39637829p1DH9y&id=535966737504

Can someone who knows the language please translate the "conditions"? I suppose there is a NDA which should be signed before getting this boards. Also it might be helpful just to ask them on their special email-address: support_hf@ingenic.com

Greatings
Elias

sogood007, 2017-11-01:
My understanding is that you can buy the development board but you need to sign the NDA for getting documentation.   However, it appears the board only ship within China. (from the taobao site)

Kadalia, 2017-11-01:
Hello,
I don't know what exactly you are looking for .... but did you try the ftp site of ingenic ?
ftp://ftp.ingenic.com/DevSupport/Tools/Toolchain/ for instance.
I do not have find the SDK, but perhaps there are interesting things
Philippe

PischeDev, 2017-11-01:
I already searched everywhere, nothing about the T20.

ykhandler, 2017-11-01:
We're looking for SDK that might contains RTSPSVR code that we can compile and use...

manjerico, 2017-11-01:
Are they obligated to share the SDK? (sorry, noob question)

Also, has anyone sent and email to the address found by @EliasKotlyar ?

[171 more replies]