samtap/fang-hacks #235: Support Chuangmi 720p camera?

jymbob, 2017-10-13:
Just picked up a pair of "Chuangmi" cameras from GearBest - https://www.gearbest.com/ip-cameras/pp_701750.html

They're using the same app, and appear to be very similar in setup.

My FTDI is in the post so I may be able to provide some debugging information if that's helpful.

VINCE43, 2017-10-17:
Hi, 
I just try tonight, and it doesn't work .. :-(

bernardo-amaral, 2017-10-31:
I wanna hack this model too

chris777c, 2017-11-06:
I have this model too, I can help you if you need anything tell me

samtap, 2017-11-06:
- Post teardown pictures
- Intercept firmware update
- Attach serial console to capture boot logs, dump filesystem, crack root passwd etc.
- Find SDK, code examples etc.

chris777c, 2017-11-06:
OK I undesstand I am so enthusiastic, in my level :

- Post teardown pictures ==> OK, I can do that

- Intercept firmware update ==> How to do that ? with what tool ?

jymbob, 2017-11-06:
I'll see if I can get anything out of the serial console this evening

chris777c, 2017-11-06:
Ok thank you

jymbob, 2017-11-06:
Hm. Think I've identified the RX and TX pins but my soldering's appalling, so I haven't got to a console yet.
It looks like it's using the same SoC as the Mijia 360 720p, which [this project](https://github.com/Filipowicz251/mijia-1080P-hacks) has working, however I couldn't get ssh access working on mine, and reflashing the 360 firmware appears to have left me with a non-functional camera - I still have a working one, so I'll persevere for a bit longer!

chris777c, 2017-11-07:
Thank you for you work

rap1, 2017-11-29:
Hi,
I also have  Chuangmi 720p camera  and would like to make it work locally, for example through RTSP, without Internet connection. 
Do you have any update?
thank you

ras31415, 2017-11-30:
Hi;
I also have 3 pcs :-(
I was little bit stupid and I didn't think that it is a closed system without open streaming protocolls.
Thanks before

rap1, 2017-11-30:
the same for me: I found out that it works only through Internet.
I hope that, as already done for other Xiaomi cameras, it's possible to load some kind of RTSP server to stream the video locally. I look forward to receiving some good news...

LarryN3t, 2017-12-12:
Serial interface connected
MAIN CHIP GM8135S
Wireless Chip realtek 8189

need support to develop img

rap1, 2018-01-08:
Hi Larry, any news?
How did you succeed in connecting serial interface ? 
thank you

LarryN3t, 2018-01-08:
Yes... connected and hacked.

ras31415, 2018-01-08:
This is very joyful.
Could you share the solution of the hacking with we?

rap1, 2018-01-09:
Great !  Could you give us details ?
Is it possible to load and run a local RTSP server ?

sibojia, 2018-01-10:
@LarryN3t Could you share the solution? I also wish to connect to this camera. Thanks a lot!

scapman, 2018-01-13:
Very great news!
Could you please share the solution?
It looks like you are the only one that made it.

Thanks a lot!

lilipfem, 2018-01-15:
Please, please share your solution if one exist  :-/

This camera make me crazy ...

bernardo-amaral, 2018-01-19:
anyone had solved this?

ril3y, 2018-01-20:
I ordered one.  I will get FH working on if its possible.  However gearbest
takes a few weeks to get it.

LarryN3t, 2018-01-21:
Hi all,
sorry for delay, this is my hack procedure for chuangmi 720p:

Connect the camera by phone app, Mi home, and configure wireless connetion to your wireless network.
Your camera will be reached by cloud.

First Hack:
Look the IP address assigned to camera by your wireless router, try to telnet: 
Username root 
Password blank

enjoy  

For the recovery image i will be share you the next week, i need to check if are present my personal configurations insight.

i want to suggest you the path to complete the hack:

After login by telnet, you can copy memory camera partitions on SD Card:

# fdisk -l 

First partition is "bootloader"
Second partition is "Kernel"
Third partition is  "Filesystem"
fourth partition is "Data" 
fifth partition "Checksum"

you can create recovery image with:

cat Kernel Filesystem Data Checksum > tf_recovery.img

If you try to create the hack alone, use other xiaomi camera hack and compare your Filesystem with another hack.

"Filesystem" image can be mounted on yor PC.

See you later.

Larry

chris777c, 2018-01-21:
Thank you for feedback but I tried telnet with putty, his show me "Connection refused"

ras31415, 2018-01-21:
Yes, absolutely same happened with me too.
telnet - l root 192.168.3.101 23
telnet: unable to connect to remote host: Connection refused.

LarryN3t, 2018-01-21:
this is original image, put in sd and turn on the camera
https://drive.google.com/file/d/1xuBw8I_-rHEyBcXRPGjyBLPwiEpusU6B/edit
With this i connect by telnet

chris777c, 2018-01-21:
Thank you, it's work, I am extract img from camera.

lilipfem, 2018-01-22:
Where do you take the .img ? The link don't work any more  :-(

ras31415, 2018-01-22:
Where did LarryN3t disappear?
His img is working.

lilipfem, 2018-01-22:
Si please, where can I found it ? I'm a poor newbies

ras31415, 2018-01-22:
This is a link where you can download larryN3t's image for 24 hours. Unfortunetelly my dropbox is full so I can't keep it there.

https://www.dropbox.com/s/qm4squv21ik4bob/tf_recovery.img?dl=0

lilipfem, 2018-01-22:
Thank you very much

jymbob, 2018-01-22:
Did a quick fork and stuck larryN3t's file [here](https://github.com/jymbob/fang-hacks) - untested.

rap1, 2018-01-23:
Could you please explain how to connect to Chuangmi camera and load the new image ?
Thank you very much

rap1, 2018-01-23:
@jymbob when you say "For Chuangmi 720 camera download tf_recovery.img, transfer to an SD card and boot" ,  do you mean to copy " tf_recovery.img" file on the SD without uncompressing  it  , correct? 
Should we prepare an sd-card with two partitions as explained below  or is not needed ?
Could you please clarify?
Thank you very much

jymbob, 2018-01-23:
I haven't actually tried it yet, just repeating what the OP wrote here. I
will endeavour to try it and update the readme. As I said, my main concern
was making sure the image was accessible somewhere.

If anyone has any feedback, I'm happy to update the readme.

lilipfem, 2018-01-23:
Try 
https://sourceforge.net/projects/win32diskimag to er/files/latest/download
To copy the .img to the SD card
Choose the file, the destination, and click write
I'm trying to see how tell the cam to boot to the SDcard

rap1, 2018-01-23:
Thank you lilipfem ! Let us know if you succeed in booting from the SD card

lilipfem, 2018-01-23:
No it's does'nt work. It's not a boutable image  :-(

jymbob, 2018-01-23:
Found some time.
You don't need to unpack the image. Simply put it on the root of an SD
card. The camera checks for this file on bootup

Steps to get telnet access:
1. Set up camera with MiHome app
2. Go into details and make a note of camera IP address (e.g. 192.168.1.250)
3. Copy tf_recovery.img to the root of an SD card
4. Unplug camera
5. Insert SD card
6. Plug in camera
7. wait for about 2 minutes until lights go blue
8. telnet 192.168.1.250 (same IP as in step 2) username: root, no password

Given we now have root access, we should hopefully be able to run a local
RTSP server.

rap1, 2018-01-23:
Great news !  Thank you very much for sharing the procedure :)
Maybe the RTSP server is already present in the image  (url: rtsp://device-ip/unicast) , could it be ?

lilipfem, 2018-01-23:
Thank you @jymbob for your help.
I can now open tenet on the cam and that's all  :-(
i answer root for the login and after only a # prompt
Everywhere i saw that i would have to connect with a url like http://192.168.1.19/cgi-bin/status

Else i try with VideoStation of Synology and vlc and nothing work

rap1, 2018-01-24:
..how did you format the SD ?  FAT16, FAT32 or other ?

lilipfem, 2018-01-24:
If the question is for me, FAT32 ...

rap1, 2018-01-24:
I also have a FAT32 formatted SD and now I can also open telnet on the cam :)
Now we should understand how to run a local RTSP server..

joshdinsdale, 2018-01-26:
Following this with anticipation! I have gotten telnet working, whats involved in getting an rtsp server running, i know a bit of linux stuff but not so much about these prebuilt firmware images. Happy to help if anyone can point me in the right direction.

tomsteenbakkers, 2018-01-28:
I’m also stuck after i got telnet access. I format a SD card with FAT32 and but the image tf_recovery.img on it. Switch off the camera. Put the SD card in the camera and power it on. When the blue light stopped blinking I checked with Fing and now the telnet protocol is there on port 23. 
Next I removed the card, and downloaded the Image to write directly to a sd-card using Win32 Disk Imager. The I removed the power from the camera, but the power back on and when the blue light stopped blinking I inserted the card. But when I used http://192.168.1.9/cgi-bin/status nothing happens. Also http://192.168.1.9/cgi-bin/hello.cgi is not doing anything. However, I can make a telnet connection. Is there a way to activate the hack using telnet? Or I’m doing something wrong. The current firmware on my camera is 3.3.6_2017092810 and the camera is still working using the app

jymbob, 2018-01-29:
The fang hack is designed for different firmware. I'm talking with the guys
who hacked the mijia 1080 which is similar firmware. It looks like we need
to recompile the rtsp library against the SoC SDK. I've had some success
getting a few things running manually on the camera (ssh, httpd) but we're
still a way off a nice turnkey solution. There are a lot of parts to the
process which are new to me, and I've only got limited time available. If
anyone has experience creating uboot images that'd be a great help.

rap1, 2018-01-31:
I noticed that on Fang hacks, they use "snx_rtsp_server", we could try to use the same executable and see if it runs,  or if we need to recompile it , where can we find the SoC SDK ?
@jymbob did you have any news from the guys who hacked the mijia 1080 ?

jymbob, 2018-01-31:
I've found what looks like an SDK for the Grain Media chipset (I don't know any Chinese). A search for "gm813x sdk" should give you some download links. The rtsp server for the mijia 1080p is [here](https://github.com/willthrom/rtspServer_gm)

lilipfem, 2018-02-09:
Is there any news ? A solution founded ?

rap1, 2018-02-20:
not yet, has anyone found a solution for loading RTSP server ?

kuchy, 2018-03-18:
Hi any progress with RTSP server?

scapman, 2018-03-26:
Hi. Any good news?

alexzip87, 2018-03-30:
no news bad news :(

jymbob, 2018-03-30:
No news no news. Sorry, no time to spend on this at the moment.

sdesalve, 2018-08-23:
Hi @jymbob, is there any news? 

Have you founded time to run RTSP server on our Chuangmi Cam?

sdesalve, 2018-08-23:
@zloybatanik 
not worked for me
My Chunagmi firmware is 3.3.6_2017092810

Mi Home suggest me an UPDATE: 3.3.6_201802014. Should I apply them?

zloybatanik, 2018-08-24:
@sdesalve  try update 3.3.6_201802014 first

sdesalve, 2018-08-24:
@zloybatanik doesn't work for me.

I've used recovery image (https://github.com/jymbob/fang-hacks) that downgrade firmware to 3.3.6_2017092810 and let me login thru telnet.

Then I've installed this https://github.com/ghoost82/mijia-720p-hack but without lucky: is there only a telnet server and mijia-720p-hack.cfg is ignored...

I've also tried to upgrade firmware to 3.3.6_2017092810 with MiHome but same result: don't work 
:(

[35 more replies]

closed 2018-09-03 by jymbob