samtap/fang-hacks #236: Some basic questions for the Xiaomi, please help me troubleshoot

DejayRezme, 2017-10-13:
My symptoms:

- MAC 34-CE, physical setup button
- Upgraded to 3.2, then downgraded using Yk Handler's firmware again.
- Camera speaks, confirms it's connected to wifi and gets assigned an IP
- On boot blinks yellow, then yellow / blue, then blinks blue. Never solid blue.
- But camera always times out in MiFi "Get camera information failed, please check network connection" (but I can see that it's firmware 3.0.... again
- Pinging the local IP (192.168.2.102) doesn't work
- http://192.168.2.102/cgi-bin/hello.cgi can't be reached (not 404 but "This site can’t be reached 192.168.2.102 took too long to respond.")
- Can't set "AP mode" (it's set to remote connection now, but don't see a different way of adding the camera

Is the LED supposed to turn solid blue once it's connected properly? Is that my underlying problem?

samtap, 2017-10-13:
I don't know Yk Handler's firmware so I can't support it

closed 2017-10-13 by samtap

DejayRezme, 2017-10-13:
Sorry @samtap I guess it's not really a legitimate issue. I just don't know where to ask and I'm really frustrated with this. Could you please please please just answer these two questions? (just a yes/no would be great. Cheers.)

With the original firmware, is it supposed to ping back to a ping command? (mine never did that)
Is it supposed to stay blue when it's connected to wifi?

samtap, 2017-10-13:
I recommend to setup a network connection using Mi Home. Regardless if you can watch the stream (blocked outside china), you can still set it up using the QR code and view the assigned ip in General Settings. There are other ways, but this is the most fool proof. After scanning the QR code and it speaks, wait a minute or so before continuing to the next step, sometimes it helps prevent the timeout. If it still times out just repeat (reboot camera). In my experience it just times out sometimes for no reason, keep trying until it works. You may notice it gets an IP and you can access everything and it still times out in Mi Home. Just try again until it doesn't.
You should then be able to ping and connect to the camera webpage. The 'index.html' is an empty page (not a unable to connect etc.). The cgi-bin/hello.cgi should have some text or at least a window title (depending on fw version and sd-card contents). You can do all of this without having sdcard inserted, when you get to this point reboot once or twice to make sure it keeps a network connection, before starting to mess with fang hacks.
To apply the hack, you need to be on a firmware version that still runs snx_autorun.sh from the sd-card (anything before v3.2.0.30). After checking you can access the webpage, insert the sd-card. You should hear the ding-ding sound and then should be able to access status page.

Disregard the LED it's unreliable and acts different depending on fw version. After applying the hack (do another reboot to make sure it sticks). you can upgrade to v3.2.0.30, or stick with the older version.

DejayRezme, 2017-10-13:
Thank you! It seems I have a network problem since ping never worked, even though I added it to Mi Home and it has an IP (before I even used an sd card or changed firmware). 
So I'll try changing some things with my wlan router or maybe disabling it completely. I've also ordered a usb TTL serial cable. Too bad these can't be setup using usb.

ykhandler, 2017-10-14:
My firmware is containing only root and etc partitions. So, if there is an error in the kernel, you must use the firmware that Samtap already shared. Then after that you can use mine. 
FYI, the original firmware contains: Uboot, kernel, rescue and root partitions.

ykhandler, 2017-10-19:
Still not enough space although snx_rtsp_server compressed with upx...

samtap, 2017-10-19:
@ykhandler Could you create a firmware that only puts some files in /etc? Just the modified startup scripts to start stuff from sd-card, so I wouldn't need to rely on the old firmware with snx_autorun.sh. 
I guess if you remove iCamera (and lots of other stuff) you'll have plenty of room. I am planning on some kind of 'app2sd' support to run stuff without sdcard, switching out iCamera with rtsp server etc.

ykhandler, 2017-10-19:
Yes i can. Please send me the script. I'll post it here at night.

ykhandler, 2017-10-19:
https://www.dropbox.com/s/89nb99sx5lz0ol4/FIRMWARE_660R.bin?dl=0

Try this, 
The script must be named snx_handler_run in the sdcard. Also you can backup all the partitions if you create a filder named HandlerBackup in the sdcard.

ykhandler, 2017-10-20:
I'm still got a lot of things to do right now. Will create a tutorial and hope your support in my youtube channel later. 

Best regards

Handler

ykhandler, 2017-10-23:
https://youtu.be/JXJfJ24rjuw

samtap, 2017-10-23:
Nice trick ;-). But if you use the proper tools to generate a fw image, I assume this is all done for you automatically?

ykhandler, 2017-10-23:
That's right. But at first i use this trick when i change the root partition. Manually edit the FIRMWARE_660R.bin. And i actually not creating any proper tools when generating the FIRMWARE, i just interrupt the file that used when generating the Firmware. Just see my next tutorial. 
Got my phone bricked today, the tutorial postponed.