samtap/fang-hacks #237: Manual setting of IP address

zebulon30, 2017-10-17:
How can I set the IP address manually ?
What file to modify ?
Thanks

zebulon30, 2017-10-17:
Good Idea, but I don't know how to do that. My router use dd-wrt software.
Did you know how to configure or if it's possible with it ?

ale2999, 2017-10-17:
https://www.dd-wrt.com/wiki/index.php/Static_DHCP

Look at the configuration piece

zebulon30, 2017-10-18:
Hi, thank you very much.
I didn't saw this previously. That's grenat .
But I try it and doesn't seems to work.
However, I entered the mac address and the host name and the final ip I wanted of course ...
I will investigate more.
Many thanks again to point me to this usefull tool.

Have a nice day.
Guillaume

⁣Envoyé par BlueMail ​

yurir, 2017-10-18:
If I am not mistaken, there is a config file on SD card of the camera, where you can enter static IP address.

yurir, 2017-10-18:
Read this discussion: https://github.com/samtap/fang-hacks/issues/60

zebulon30, 2017-10-18:
Yes, that's what i was thinking about.
But my try give up and hangs the camera and i had to make all again (reseting the camera, fang hack, image.jpg hack, ...)
So if someone know the exact procedure, i will follow it ...
Thanks

⁣Envoyé par BlueMail ​

zebulon30, 2017-10-19:
Thans to you.
I will give it a try.

⁣Envoyé par BlueMail ​