samtap/fang-hacks #238: SD card not found from MiHome (3 different SDs)

johnickz, 2017-10-18:
Hello and sorry if this a known issue but my camera cant see three different SD cards from MiHome (Kingston 8gb, Samsung 32gb, Sandisk 32gb). Formatted to FAT32 but MiHome cant see both of three SD cards. The hack is working fine if I prepare both of these SDs and insert them before the boot of the camera. 

Any info will be appreciated, thank you.

ykhandler, 2017-10-18:
Use the original cable, also adaptor and try to change your adapter (from adaptor to power source). Usually this happen because of insufficient power...

johnickz, 2017-10-18:
I just used original cable and two different power adaptors (one 1000mA and one 2000mA) nothing happens. SD card not found on MiHome.

samtap, 2017-10-18:
The sdcard script that is copied by fang-hacks isn't compatible with fw V3.2.0.30, which is why the sdcard isn't listed in Mi Home, but works fine in all other regards. If you want to use it from mi home you'll have to downgrade, or replace the sdcard script manually.

This will be fixed in a future release.

johnickz, 2017-10-18:
Thank you for your fast answer and for your great effort for all us owners. 

Is this easy to do to replace the script or downgrade? Any link if possible?
Sorry to bother but do you have any time schedule in order to wait for your next release?

samtap, 2017-10-18:
No schedule but it will happen, I've not given up on this device yet ;-).

Possibly you can find the script in /root/etc_default/hotplug/sdcard but I'm not sure and don't have access to a cam right now. To downgrade you should follow instructions here: https://github.com/samtap/fang-hacks/wiki/HowTo:-Flash-original-Xiaomi-firmware-from-sdcard-(factory-reset)#via-sdcard

johnickz, 2017-10-18:
Glad that I hear that you are still standing Sir!

Thank you for your useful info.
You can lock this thread.

mouth4war, 2017-11-07:
+1 for this issue. I thought I needed to resize partitions but not a single SD card works. This issue prevents turning on continuous recording via Mi Home app.

canuyanik, 2017-11-16:
are you hearing any bell sound when you put sd card in camera?

johnickz, 2018-05-30:
Update.

I downgraded easily with the Guide, at first 3.0 version no SD card found. 
After updating to 3.3, no SD card again found.
Then it notified me about 3.4 version update, did it and now SD card works like a charm!!

And yes, now I am hearing a two bell sound when I am inserting my SD!