samtap/fang-hacks #239: Unable to start rtsp stays NOK

dendrees, 2017-10-19:
I have 4 times applied the hack step for step. Started all over with a new sd card. But I am still struggling to start the rtsp server. I always stays NOK. I have read the FAQ telling me to disable the cloud service.
When I hit 00-stop cloud I get the message 

Running script '00-stop-cloud'...
Stopping Cloud applications...
sh: you need to specify whom to kill
sh: you need to specify whom to kill
sh: you need to specify whom to kill

The proces list is showing me that the cloud service are still running.
619 root     10772 S    /etc/app/iCamera
  628 root     64728 S    /etc/app/iCamera

I don't have a clue anymore?

[image]
[image]
[image]

dendrees, 2017-10-19:
By the way this is my log:

Contents of /tmp/hacks.log:
Thu Oct 19 15:30:00 GMT 2017 - /etc/fang_hacks.sh: Removed stale logfile
Thu Oct 19 15:30:00 GMT 2017 - /etc/fang_hacks.sh: Executing script (enabled: 1)
Thu Oct 19 15:30:00 GMT 2017 - /etc/fang_hacks.sh: Linking /media/mmcblk0p1/bootstrap/www/action -> /tmp/www/cgi-bin/action
Thu Oct 19 15:30:00 GMT 2017 - /etc/fang_hacks.sh: Linking /media/mmcblk0p1/bootstrap/www/action.orig -> /tmp/www/cgi-bin/action.orig
Thu Oct 19 15:30:01 GMT 2017 - /etc/fang_hacks.sh: Linking /media/mmcblk0p1/bootstrap/www/func.cgi -> /tmp/www/cgi-bin/func.cgi
Thu Oct 19 15:30:01 GMT 2017 - /etc/fang_hacks.sh: Linking /media/mmcblk0p1/bootstrap/www/scripts -> /tmp/www/cgi-bin/scripts
Thu Oct 19 15:30:01 GMT 2017 - /etc/fang_hacks.sh: Linking /media/mmcblk0p1/bootstrap/www/status -> /tmp/www/cgi-bin/status
Thu Oct 19 15:30:01 GMT 2017 - /etc/fang_hacks.sh: Running startup scripts
rmmod: remove 'snx_wdt': Resource temporarily unavailable
Thu Oct 19 19:36:31 GMT 2017 - /etc/fang_hacks.sh: Finished
rmmod: remove 'snx_wdt': Resource temporarily unavailable
rmmod: remove 'snx_wdt': Resource temporarily unavailable

dendrees, 2017-10-20:
Anyone?

miki-inf, 2017-10-28:
had the same issue switched to main  screen 
reapply the script than rebooted the system and it worked .

dendrees, 2017-10-30:
@miki-inf I tried that option already in my situation. Then I get a message like nothing has been changed or script was already installed. I found out that the cloud processes are preventing me from starting the Rtsp stream.

dendrees, 2017-10-30:
Is there a way to stop the cloud process manually?

samtap, 2017-10-31:
You can try running the stop-cloud script manually. The 'rmmod: remove 'snx_wdt': Resource temporarily unavailable' indicates its trying to unload the module while cloud apps are still using it. Strange behaviour that I haven't seen before. 
You should login with telnet/ssh, run /media/mmcblk0p2/data/etc/scripts/00-stop-cloud and check the output. If you need help please join gitter chat and I can give some more tips.

trytek235, 2017-11-02:
How is login telnet?

trytek235, 2017-11-04:
I have the same problem. I have inserted two diffrent sd card. When I click to stop clouds app, this appear.
sh: you need to specify whom to kill
sh: you need to specify whom to kill
sh: you need to specify whom to kill
Pls help.

Contents of /tmp/hacks.log:
Thu Jan  1 08:00:05 GMT 1970 - /etc/fang_hacks.sh: Executing script (enabled: 1)
Thu Jan  1 08:00:05 GMT 1970 - /etc/fang_hacks.sh: Waiting for cloud apps...
Thu Jan  1 08:00:06 GMT 1970 - /media/mmcblk0p1/snx_autorun.sh: Running (device: mmcblk0p1)
Thu Jan  1 08:00:06 GMT 1970 - /media/mmcblk0p1/snx_autorun.sh: Created /tmp/www/cgi-bin
Thu Jan  1 08:00:11 GMT 1970 - /etc/fang_hacks.sh: iCamera is running!
Sat Nov  4 21:40:19 GMT 2017 - /etc/fang_hacks.sh: Boa webserver is running!
Sat Nov  4 21:40:19 GMT 2017 - /etc/fang_hacks.sh: Linking /media/mmcblk0p1/bootstrap/www/action -> /tmp/www/cgi-bin/action
Sat Nov  4 21:41:04 GMT 2017 - /etc/fang_hacks.sh: Linking /media/mmcblk0p1/bootstrap/www/func.cgi -> /tmp/www/cgi-bin/func.cgi
Sat Nov  4 21:41:04 GMT 2017 - /etc/fang_hacks.sh: Linking /media/mmcblk0p1/bootstrap/www/network -> /tmp/www/cgi-bin/network
Sat Nov  4 21:41:04 GMT 2017 - /etc/fang_hacks.sh: Linking /media/mmcblk0p1/bootstrap/www/scripts -> /tmp/www/cgi-bin/scripts
Sat Nov  4 21:41:04 GMT 2017 - /etc/fang_hacks.sh: Linking /media/mmcblk0p1/bootstrap/www/status -> /tmp/www/cgi-bin/status
Sat Nov  4 21:41:05 GMT 2017 - /etc/fang_hacks.sh: Patch command not found! Patches will not be applied.
Sat Nov  4 21:41:05 GMT 2017 - /etc/fang_hacks.sh: Running startup scripts
rmmod: remove 'snx_wdt': Resource temporarily unavailable
Sat Nov  4 21:41:17 GMT 2017 - /etc/fang_hacks.sh: Finished
/media/mmcblk0p2/data/etc/scripts/00-stop-cloud: line 8: rmmod: not found

samtap, 2017-11-07:
@dendrees Is your issue resolved?

dendrees, 2017-11-08:
Not yet. I will try with SSH later this week. How can I enable the cloud proces manually? Just in case of.
Side question when I disconnect the power from the device I need to insert the SD-card till I hear the hack sound. Otherwise I am not able to login to the gui. Is the cloud process preventing this as well?

samtap, 2017-11-08:
Just trust the scripts, they work. Don't fiddle with manual stuff unless you know what you're doing. You need to leave the sdcard inserted to use fang-hacks.

dendrees, 2017-11-08:
It's not working in my case. The hack is not permanently enabled when I reconnect the device. I need to insert the sd card again to get into the gui.

When I click to stop clouds app in the gui, this appear.
sh: you need to specify whom to kill
Same issue mention here above by @ttrytek235
I am really having a hard time to get it working.
I will try your workaround to stop the cloud process in ssh.

samtap, 2017-11-08:
No idea what you're doing wrong. Did you even apply the hack using the webif the first time the card was inserted?

trytek235, 2017-11-08:
I resolve the issue using terminal command Linux instead of Windows disk imager. I used Linux on VirtualBox on Windows. I format via Gnome-disks and use terminal to use image on SD card. And now works. Sorry for bad English.

dendrees, 2017-11-13:
@samtap yes everything went fine. I applied the hack multiple time. Just the cloud part doesn't work.
@trytek235  Thx for the info. I don't have linux but I will try with Easus Partion manager/ Etcher.

samtap, 2017-11-13:
@dendrees Sorry I'm lost. If what you're saying is true, your device is broken since it's impossible to apply the hack multiple times, unless the internal flash is broken and doesn't store any changes.

dendrees, 2017-11-14:
@samtap I mean just the hole procedure from start and preparing the sd card. 
Update: I have successfully flashed 1 of my 2 camera's. And the hack is working fully.
I have deleted alle the two partitions, created a new single partition and formatted the drive with Easus Partion manager. Instead of using Win32 Disk Imager I used "Etcher" to flash the SD-Card image, V0.2.0.
Everything is green now! Maybe there is a problem with W32 Disk imager? I will try the second camera this week.

adrianmihalko, 2018-03-24:
I am using etcher, but I have same problem.

```
Contents of /tmp/hacks.log:
Thu Jan  1 01:00:05 CET 1970 - /etc/fang_hacks.sh: Executing script (enabled: 1)
Thu Jan  1 01:00:05 CET 1970 - /etc/fang_hacks.sh: Waiting for cloud apps...
Thu Jan  1 01:00:06 CET 1970 - /media/mmcblk0p1/snx_autorun.sh: Running (device: mmcblk0p1)
Thu Jan  1 01:00:06 CET 1970 - /media/mmcblk0p1/snx_autorun.sh: Created /tmp/www/cgi-bin
Thu Jan  1 01:00:10 CET 1970 - /etc/fang_hacks.sh: iCamera is running!
Mon Oct 10 18:00:28 CEST 2016 - /etc/fang_hacks.sh: Boa webserver is running!
Mon Oct 10 18:00:28 CEST 2016 - /etc/fang_hacks.sh: Linking /media/mmcblk0p1/bootstrap/www/action -> /tmp/www/cgi-bin/action
Mon Oct 10 18:00:29 CEST 2016 - /etc/fang_hacks.sh: Linking /media/mmcblk0p1/bootstrap/www/func.cgi -> /tmp/www/cgi-bin/func.cgi
Mon Oct 10 18:00:29 CEST 2016 - /etc/fang_hacks.sh: Linking /media/mmcblk0p1/bootstrap/www/network -> /tmp/www/cgi-bin/network
Mon Oct 10 18:00:29 CEST 2016 - /etc/fang_hacks.sh: Linking /media/mmcblk0p1/bootstrap/www/scripts -> /tmp/www/cgi-bin/scripts
Mon Oct 10 18:00:29 CEST 2016 - /etc/fang_hacks.sh: Linking /media/mmcblk0p1/bootstrap/www/status -> /tmp/www/cgi-bin/status
Mon Oct 10 18:00:29 CEST 2016 - /etc/fang_hacks.sh: Running startup scripts
modprobe: remove 'snx_wdt': Resource temporarily unavailable
Sat Mar 24 23:41:05 CET 2018 - /etc/fang_hacks.sh: Finished
modprobe: remove 'snx_wdt': Resource temporarily unavailable
```
```

~ # /media/mmcblk0p2/data/etc/scripts/00-stop-cloud
Stopping Cloud applications...
Module snx_wdt unloaded (0)
~ #
fi~ #  modprobe -rf snx_wdt
modprobe: remove 'snx_wdt': Resource temporarily unavailable
```

adrianmihalko, 2018-03-24:
I have latest 3.3.3.25 firmware on my camera.

adrianmihalko, 2018-03-24:
Running processes:

```
~ # ps
  PID USER       VSZ STAT COMMAND
    1 root      1164 S    init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [events/0]
    5 root         0 SW   [khelper]
    8 root         0 SW   [async/mgr]
  209 root         0 SW   [sync_supers]
  211 root         0 SW   [bdi-default]
  213 root         0 SW   [kblockd/0]
  223 root         0 SW   [khubd]
  226 root         0 SW   [kseriod]
  231 root         0 SW   [kmmcd]
  241 root         0 SW   [cfg80211]
  262 root         0 SW   [rpciod/0]
  269 root         0 SW   [khungtaskd]
  270 root         0 SW   [kswapd0]
  316 root         0 SW   [aio/0]
  323 root         0 SW   [nfsiod]
  332 root         0 SW   [crypto/0]
  399 root         0 SW   [mtdblock0]
  404 root         0 SW   [mtdblock1]
  409 root         0 SW   [mtdblock2]
  414 root         0 SW   [mtdblock3]
  419 root         0 SW   [mtdblock4]
  424 root         0 SW   [mtdblock5]
  429 root         0 SW   [snx-spi.0]
  438 root         0 SW   [zd1211rw]
  485 root         0 SW   [usbhid_resumer]
  518 root         0 SWN  [jffs2_gcd_mtd4]
  569 root         0 SW   [isp]
  593 root         0 SW   [mmcqd]
  682 root         0 SW   [flush-179:0]
  710 root      1160 S    /sbin/getty -L ttyS0 115200 vt100
  711 root      1152 S    /sbin/syslogd -n -m 0
  712 root      1148 S    /sbin/klogd -n
  763 root         0 SW   [flush-mtd-unmap]
  774 root         0 SW   [RTW_CMD_THREAD]
  885 root      1088 S    wpa_supplicant -Dwext -iwlan0 -c/tmp/wpa_supplicant.conf -B
  919 root      1164 S    udhcpc -i wlan0 -p /var/run/udhcpc.pid -b
  937 root       760 S    /usr/boa/boa
 1006 root      1352 S    ntpd -p time.google.com
 1010 root      1360 S    telnetd -p 2323 -l fang_login
 1016 root      1348 S    tcpsvd -E 0.0.0.0 21 ftpd -w /
 1018 root       572 S    dropbear -R -P /var/run/dropbear.pid
 1019 root     17668 S    snx_rtsp_server -W 1920 -H 1080 -Q 10 -b 4096 -a
 1021 root      1152 S    {fang-ir-control} /bin/sh /media/mmcblk0p2/data/usr/bin/fang-ir-control.sh
 1040 root      1148 S    {fang_login} /bin/sh /media/mmcblk0p2/data/bin/fang_login
 1041 root      1160 S    -sh
 1092 root      1344 S    sleep 3
 1093 root      1352 R    ps
```