samtap/fang-hacks #242: Help Please; Almost there with fang hacks..issues with cam sleeping.

pointbob, 2017-10-23:
First off thanks for your patience; I'll try to be brief.

I applied the fanghacks_v0.2.0 hack and the led is blue/connected to my wifi/router.

Issues:

1) While view the rtsp thru VLC or tinycam pro in LIVE mode; the camera image goes black/sleep and I have to restart it by pressing play (on VLC). How do I make the camera to stream live without going to sleep? I've attached screen caps of my hack settings.

2) How exactly do I add a script? I mean ....where exactly do i enter the parameters? Is it on the http://192.168.1.xxx/cgi-bin/status   status page somewhere/?? pics attached.

3) Some of the boot scripts say NOK (in red)...is this a bad thing? how do I make it all green and is that necessary? pic attached

4) Networks settings on hack page show wireless mode is set on CLOUD. Other options are Wireless Client and Access Point. I mainly want the camera to be used at home and for me to access the cam stream from at work. Is cloud the way to go for this or should I choose wireless client?

5) Motion detection. Is there anyway to get the motion detect working via camera? It sort of works with tiny cam but that has to be on 24/7 with tinycam monitoring it live to access if motion occurs. Any workaround here or is the hack incapable of accessing the motion detect on the camera? Interestingly the Mi Home app shows the cam as OFFLINE since the hack was applied. I guess they cannot function together....?

THANKS for any help...great community
[image]

[image]

[image]

samtap, 2017-10-25:
1. No idea what could cause it. It runs stable for me, did you change any rtsp server settings? Also check the VLC debug logs for clues.
2. Adding a new script means creating a file in /media/mmcblk0p2/data/etc/scripts. You can use ssh, ftp or just take out the sd-card to copy the file using a PC. Check the other scripts for style/format, you can have a script that runs once, i.e. like the 00-stop-cloud script, or a script with start/stop/status functions like the others. It will show up on the scripts web-page automatically.
3. They're supposed to be green after booting, indicating all services are started succesfully. Run scripts manually from SSH to check for errors.
4. Cloud means network is untouched and iCamera (the Xiaomi application) manages the network, i.e. you pair it with mi home scanning the QR code. This is fool-proof and safe since you can always repeat the pairing process. But it requires iCamera to start and setup the network which takes a long time. Client-mode wifi means fang-hacks scripts setup the network thus iCamera can be disabled and the device boots fast. But if it doesn't work, there's no easy way to fix it since you will not be able to access it. There ways to recover but you have to modify files on the sdcard offline (on a pc).
5. There is support for motion detection, but you can't do anything useful like record a video-fragment or take a snapshot, yet. It will be supported in a future release. For now you can send events on motion, i.e. trigger a NVR to start recording the RTSP stream, send an email etc. but you will have to write a script for that yourself.
6. You will have to make the RTSP stream available outside your LAN. I suggest a VPN or SSH tunnel for security, or waiting for a future release that supports authentication. You can just port-forward port 554 but anybody will be able to view your camera that way.

pointbob, 2017-10-26:
Outstanding replies and thank you very much.