samtap/fang-hacks #243: Wyze Cam firmware to Xiaofang?

kulokoy, 2017-10-25:
Hi just wondering if its possible to use wyze cam firmware to xiaofang camera? I've seen the demo of the product and its literally same as xiaofang except for the app which is modified mi home in english version which is nice and working well.

https://www.wyzecam.com/product/wyze-cam/

Its only available in US so american who bought xiaofang cam before are lucky that they can buy a proper working security camera without problem in security and geo blocking. Hope somebody can port wyze FW to XioFang Cam.

samtap, 2017-10-25:
I'm sure it runs as it's the same hardware but they will likely block device-id if it's not sold by them (same as iSmartAlarm). 

I've contacted them to ask about open-source community/support. It would be nice to join forces, as they do sell a device that makes extensive use of open-source software (Linux) already.

kulokoy, 2017-10-25:
That's what i thought, it could be the same as the ismartalarm that they selling here in europe but 3x higher price than the wyze cam. Hopefully someone could port it to xiaofang. As what you said its better if they will provide open source code of the FW..

tomsbrinkis, 2017-10-31:
I really hope, someone could make this happen - port Wyze firmware to xiaomi cameras, becaus i have 3 cameras at home that does not work.

samtap, 2017-10-31:
According to Wyze they're using the same design, but different components inside. Someone should do a teardown guide with pics so we can check the differences. I doubt there're much differences, perhaps a different wifi chip?

jat255, 2017-11-01:
Hi everyone, I was previously unfamiliar with this project, but I scooped up one of the Wyze cams and it arrived last night. I disassembled it to see what was inside, and took some photos of the journey. Hopefully they can be of some help in figuring out if this camera will be compatible. The photos are in an [imgur album](https://imgur.com/a/dvPxs). Let me know if you have any questions on the hardware side!

samtap, 2017-11-01:
Thanks, looks identical and even has the same PCB markings like iSC which clearly refers to iSmartAlarm Cam or whatever..
Would you happen to have a ftdi usb adapter to connect to the GND, TX, RX visible here: https://i.imgur.com/s4LAp5V.jpg (you don't even need to solder wires if you can get them through the holes, a c-grid header works great)
That would allow to capture the boot log and check for fw differences. I don't believe the price of only 20 bucks allows for any research/development, so it's likely just config stuff to use their cloud/app instead of Xiaomi/iSmartAlarm.

jat255, 2017-11-01:
Unfortunately, I don't... I've never done any sort of bare PCB/firmware hacking type stuff, so even taking apart the camera was a little bit of a learning experience.

I agree with you though, unless they have some sort of massive VC funding, I can't see how they would rewrite a bunch of code for this with such a low price point.

jat255, 2017-11-01:
Actually, I forgot I have a couple of these: https://docs.getchip.com/chip.html#pin-headers

I bought them but never really got around to using them for anything. Would that work? I'd need a decent amount of hand-holding, clearly.

samtap, 2017-11-01:
It's probably doable with the device you linked, if you can flash the right software on it. You can get ready to use stuff for things like Arduino or esp8266, but I'm not familiar with the CHIP board.

jat255, 2017-11-03:
While it sounds like a fun project, I'm not sure I'll have time to get around to trying to hack it together. Hopefully someone with some more hardware abilities will come along.

wsmlby, 2017-11-05:
is it possible to run this hack on the Wyzecam hardware? I push prepared the sdcard, put it in after the device boot, and I heard the sound and in "/cgi-bin/hello.cgi" i can see sdcard content("snx_autorun.sh" etc), however, the status page gives 404. 

Can you share how do you find out the "snx_autorun.sh" is what it runs at boot for fang? maybe wyzecams are using different name for the startup script?

ril3y, 2017-11-05:
Mine has shipped.  Well tear it down once I get it.

dustinsterk, 2017-11-06:
FCC submissions can be seen here:
https://fccid.io/2ANJHWYZEC1/Internal-Photos/Internal-Photos-3565884

The partner company creating them is:
http://www.hualaikeji.com/en

jat255, 2017-11-06:
@dustinsterk I guess I didn't have to tear mine apart, then... 😄

dustinsterk, 2017-11-06:
@jat255 your pictures are much clearer!  :)

samtap, 2017-11-07:
The snx_autorun.sh 'backdoor' was pretty obvious and easy to find by reading the scripts of the original XiaoFang. Because it was so easy to get basic access to the cam, figuring out how other stuff works didn't take much effort. By now, we know all the details and can just build our own firmware, that doesn't require snx_autorun.sh to kickstart things. Bits and pieces of how to do this are unfortunately scattered across lots of github issues, wiki etc, but people have been able to build their own firmware (i.e. @ykhandler). I would be surprised if you can't just flash this to a WyzeCam and make it work (wifi may require a different driver). Obviously you won't be able to connect with their cloud, but fang-hacks should work fine.

My goal isn't to build my own firmware and never has been. I prefer running stuff off the sdcard as I don't have to worry about limitations of internal flash, opening up a lot of possibilities. One of them is higher potential for re-use on other cams.

ril3y, 2017-11-09:
RIght we have the SDK's and the build scripts from the vendor but what
mechanism is there to actually flash it the device?   I am not talking
auto_run.sh trickery but camera out of box without opening it up.  Did I
miss something?  Is this known?  BTW I got my wyze last night.  Hope to
play with it this weekend.

kvcoates, 2017-11-09:
Where did you get SDK's from? which vendor

ril3y, 2017-11-09:
I found the sdk on a Chinese filesharing site like media fire.  This was
from the company that makes the SoC.  I have seen links to other sdks on
other issues.

kvcoates, 2017-11-09:
can you share the link/files with me please?

samtap, 2017-11-09:
@ril3y It runs uboot so you simply provide a firmware image on sdcard (FIRMARE_660R.bin) and it is flashed before boot. The only modification required for fang-hacks is in the boot scripts: prevent it from starting iCamera and instead run the fang-hacks.sh script (which in turn mounts and starts stuff on sd-card, if present). If you intercept a firmware update, extract the image and modify some bootscripts, add fang-hacks.sh etc, repackage it, you don't even need to build a complete fw image with the SDK.

ril3y, 2017-11-09:
@PatrickM have you verified this? The repackaging / reflashing part?  The
uboot Firmware_660.bin is specific to fang bootloader code I assume then.
Who knows what other cameras with the same hardware have compiled.
Has someone done this?@kvcoates i will look around tonight for you.

Thanks.

samtap, 2017-11-09:
@ril3y Nope not me but I've seen others like I mentioned before. The problem is with the rest of the stuff, i.e. the Xiaomi binaries won't work on a Wyze cam, you'd have to obtain an original fw image for all cams and modify it the same way. So what I really need is a image that only contains some updates for /etc and leaves the rest as is. Not sure if that can even be done from uboot. The official OTA updates are flashed using an application, not via uboot.

lokkju, 2017-11-11:
I'll just leave this here...

https://d1fk93tz4plczb.cloudfront.net/UpgradeKit/1508232228/img_wyze_snx_sys_3.9.1.42.img

kvcoates, 2017-11-11:
@lokkju Greate, except not loading? might just be having the saturday morning blues!, how did you load the ISO onto camera?

tam481, 2017-11-14:
Hello guys,
I'm slightly confused. Is the aim to get fang-hacks to work on the Wyze or to get the Wyze firmware on the Xiaomi Xiaofang in order to use Whyze's app? I'd love to put Wyze's firmware on mine and use their app but also put the SD card in with fang-hack to stream RTSP

dustinsterk, 2017-11-14:
I have dug in a little with their app/api calls.  Being able to load their firmware onto a non Wyze specific device may prove challenging.  The app calls seem to pass a bunch of metadata and MAC address of each camera along with some camera specific identifiers.  I would assume each camera also has private keys for encryption to access video on AWS.

Personally, I would rather focus on unlocking RTSP, SFTP, on the exiting Wyze hardware so that you have the ability to use their device with or without the AWS cloud integration.

samtap, 2017-11-14:
All versions of the cam (iSmart/Xiaomi/Wyze) presumably use the same mechanism to pair with cloud: some unique identification stored in nvram. You can easily clone that to a different cam, but you can't connect two cams with the same id to a cloud. One will simply get disconnected when the other one connects. Even if you're able to generate correct id's yourself, you are effectively *exploiting their cloud infrastructure*. The cloud infrastructure you don't own or pay for, but are allowed to use free of charge. By doing this, you are risking even more blocks and feature limits are put into place for the rest of us.

dustinsterk, 2017-11-14:
For $19.99 (plus shipping), honestly the camera and apps work incredibly well...Wyze did a fantastic job and I hope they continue to gain market share.

My fear is that they have undercut their pricing too far to obtain users and with the included 14 day AWS storage the company will not survive.  Enabling RTSP and other functionality will ensure these devices will remain useful should something like this happen (and I really hope it does not).

dustinsterk, 2017-11-15:
Has anyone connected to the hardware on the Rx Tx pins via serial cable?  If so, is the root password the same as xiaofang?

ril3y, 2017-11-16:
no its root ismart12

Its almost identical to the Fang app.  Clearly they used it as a base to
tune it to their needs.

Riley

samtap, 2017-11-16:
@ril3y root/ismart12 is the same as on XiaoFang. Let me know if you need help, if Wyze cam doesn't accept that.

ril3y, 2017-11-16:
@PatrickM oh I think @dustinsterk wanted to know if it was the same.  The
wyze does in fact accept that.  Its also running the "iCamera" bin vs the
iSCSI ( cant remember the other bin name)

samtap, 2017-11-16:
@dustinsterk I expect they will soon start selling other home automation/security products and a subscription service for a cloud interface to manage/monitor all of them. It's just your typical vendor lock-in scheme. Note they're ex-amazon employees so they may have a special deal on AWS, or even if they don't it's not really expensive if you set things up a certain way.

@ril3y Ok so if they didn't even take the effort / have the skills to change the root passwd, it's safe to conclude they just changed some minor cloud-connection details, switching Xiaomi for AWS. As far as I've seen, on all firmware versions the boot scripts run /etc/app/iSC3S, just on the more recent versions that binary has decreased a lot in size and is just a wrapper that in starts iCamera etc., and then terminates itself.

ril3y, 2017-11-16:
PatrickM,

No iCamera is their app they are running on the camera (WYZE).  It does not
terminate.  Its almost Identical to the iSC3S bin with some AWS stuff
changes.  I know I have been reversing them both.

samtap, 2017-11-16:
Even the naming of iCamera, iSC3S, the ismart12 passwd etc. all imply it was developed by/for the iSmartAlarm Spot. The XiaoFang runs the same. I get the impression cloud connection, pairing etc. is handled by other binaries (i.e. the miio-client stuff on XiaoFang).

ril3y, 2017-11-16:
Side note did you ever get GDB compiled for this device? Or were you able
to successfully use gdbserver and remote debug it?  I have been fighting
cross compilers for a day or 2 now.

dustinsterk, 2017-11-16:
Thanks to all, this information is very useful.

samtap, 2017-11-16:
Seems when I reply to a comment-notification email it doesn't get added to this thread? :(
@ril3y There's a 32-bit gdb build in the SDK and gdbserver is conveniently included in the firmware. So it's pretty straightforward to debug as long as you make sure to compile with debug flags (and don't strip binaries).

ril3y, 2017-11-16:
Can you link me that toolchain.  I have one but not sure its the same.  I
build gdbserver on my own.  What I have not got working is GDB built for
x86 with the ability to debug armv5 (remote).  Convoluted bit time!

dustinsterk, 2017-11-16:
After reading this thread, I would now really love to be able to disable the cloud service:
https://www.reddit.com/r/wyzecam/comments/7cykgf/wyzecam_sending_data_to_servers_other_than_aws/

wsmlby, 2017-11-17:
Shall we have a separate issue for discussing fang-hacks on WyzeCam?

dustinsterk, 2017-11-17:
I would agree, anyone else?

trevorcobb, 2017-11-22:
i would be very interested in retrofitting firmware that will enable rtsp on my wyzecam, so if that requires starting another issue to discuss, i'm all in favor.

aaronsilber, 2017-11-28:
I opened #268 for further discussion about porting fang-hacks to the Wyze devices.

Great insight so far! I spent a lot of time reversing the Android client, trying to get a stream out of the device, but with the stock firmware you are stuck using the ThroughTek P2P service.

julian-alarcon, 2017-12-01:
I have a little concern, so the root credentiasl are root/ismart12 . And the cam has SSH open by default and it's conected to Internet, so.. Anybody could login your Camera and do whatever they want!??

If this is true, is possible to change it? Also,, changing credentials will affect login of Apps or firmare updates?

ril3y, 2017-12-01:
Openssh is not installed on the WYZE.  If you have openssh running on your
WYZE then there is something very odd going on.  At least the default
version does not have it running.  What makes you think this is the case?

samtap, 2017-12-03:
Anyone know where I could order a WyzeCam, for shipping to Europe?

dariopb, 2017-12-04:
@samtap : I got a boot dump from the wysecam. I boot the cam, then I insert the hacks sd (logs at the end and I attached the file as well):

U-Boot 2011.09 (Sep 15 2017 - 20:19:00)

DRAM:  64 MiB
MMC:   SD Card not detect
mmci_host_init error - -1

SPI FLASH: 16 MB
In:    serial
Out:   serial
Err:   serial
GPIO[2] is high
Hit any key to stop autoboot:  0
roofsr size = 0x6c3070
## Booting kernel from Legacy Image at 00008000 ...
   Image Name:   Linux-2.6.35.12
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    3038112 Bytes = 2.9 MiB
   Load Address: 00008000
   Entry Point:  00008040
   Verifying Checksum ... OK
   XIP Kernel Image ... OK
OK

Starting kernel ...

Uncompressing Linux... done, booting the kernel.
Linux version 2.6.35.12 (fedora@localhost.localdomain) (gcc version 4.5.2 (SONiX                                                    GCC-4.5.2 Release 2011-12-06) ) #30 Sun May 21 23:35:25 PDT 2017
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=00057177
CPU: VIVT data cache, VIVT instruction cache
Machine: SONiX SN98600 Development Platform
Memory policy: ECC disabled, Data cache writeback
CPU: found ITCM 16k @ ffff4000, enabled
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
Kernel command line: console=ttyS0,115200 root=/dev/mtdblock2 init=/linuxrc mem=                                                   64M isp=10M vc=12M vo=0M mtdparts=snx-spi:768k(uboot),3072k(kernel),7168k(rootfs                                                   ),4096k(rescue),1024k(etc),256k(userconfig)
PID hash table entries: 256 (order: -2, 1024 bytes)
Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
Memory: 64MB = 64MB total
Memory: 36836k/36836k available, 28700k reserved, 0K highmem
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
    DMA     : 0xffa00000 - 0xffe00000   (   4 MB)
    vmalloc : 0xc4800000 - 0xe0000000   ( 440 MB)
    lowmem  : 0xc0000000 - 0xc4000000   (  64 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .init : 0xc0008000 - 0xc0026000   ( 120 kB)
      .text : 0xc0026000 - 0xc04ec000   (4888 kB)
      .data : 0xc050a000 - 0xc0535440   ( 174 kB)
SLUB: Genslabs=11, HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Hierarchical RCU implementation.
        RCU-based detection of stalled CPUs is disabled.
        Verbose stalled-CPUs detection is disabled.
NR_IRQS:96
Console: colour dummy device 80x30
console [ttyS0] enabled
Calibrating delay loop... 200.29 BogoMIPS (lpj=1001472)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 512
CPU: Testing write buffer coherency: ok
NET: Registered protocol family 16
0x00a00000 bytes system memory reserved for isp device at 0x005ec000
0x00c00000 bytes system memory reserved for vc device at 0x00fec000
bio: create slab <bio-0> at 0
SCSI subsystem initialized
usbcore: registered new interface driver usbfs
usbcore: registered new interface driver hub
usbcore: registered new device driver usb
[... 676 lines cut ...]
------------MP4 Save Head witch 1920 ----------------
------------MP4 Save Head height 1080 ----------------
joseph_mp4_config->width joseph_mp4_config->heightjoseph_mp4_config->height  1920 1080
joseph_mp4_config->width joseph_mp4_config->heightjoseph_mp4_config->height  1920 1080
joseph_mp4_config->width joseph_mp4_config->heightjoseph_mp4_config->height  1920 1080
InitMp4Encoder InitMp4EncoderInitMp4EncoderInitMp4EncoderInitMp4Encoder 111111
InitMp4Encoder InitMp4EncoderInitMp4EncoderInitMp4EncoderInitMp4Encoder end
 XXXXXXXXXXXXXXXXjoseph_mp4_config->audio XXXXXXXXXXXXX   1
[AUTO] set_auto_task_cur_state 2
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
@@@@@@@@@@@@@@@      0        6403    @@@@@@@@@@@@@@@@
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
msg_queue_rcv time: 5A24D9D5, id: 0, type: 2001, cmd: 2009
[test_UP] 01echo 1 > /proc/sys/vm/drop_caches
 XXXXXXXXXXXXXXXXjoseph_mp4_config->video XXXXXXXXXXXXX   2
login state ok 7777777777777777777777777777777 1512364527
(ISP)fps modify == 10 --> 8
ubIdx = 23, fps = 8
priv->uwMaxExpL = 4500, info->frame_rate = 8
set i frame
ubIdx = 21, fps = 10
priv->uwMaxExpL = 3600, info->frame_rate = 10
(ISP)fps modify == 8 --> 10

[wysecam_boot_serial_dump.txt](https://github.com/samtap/fang-hacks/files/1525745/wysecam_boot_serial_dump.txt)

samtap, 2017-12-04:
Nice boot log but what should I do with it? ;-) Seems I can't buy a wyzecam anywhere for a reasonable price. Though I'm pretty sure it just works the only issue is getting the scripts on the device without needing serial console.

dariopb, 2017-12-04:
I thought you wanted to check the fw differences: "That would allow to capture the boot log and check for fw differences." :)
What should I do to try to see if the scripts run on my setup? What processes should I kill and then execute the script on the sd? I can login as root with the ismart creds.

samtap, 2017-12-05:
Those are two different things ;-). Serial console allows to check the filesystem and compare it with other cams. I'd already seen the boot log and fw somewhere else so already know it's 99% identical.

The issue is kicking off the fang-hack scripts. On the Xiaomi this is handled by snx_autorun.sh which copies some stuff from sd bootstrap folder to flash, like changed boot scripts that run /etc/fang-hacks.sh automatically. So if you place a sdcard with the fang-hacks image in a wyzecam and run that script manually, it may just work. But a warning: this will overwrite some files on the device so make backups first and I highly recommend only doing this after reading the scripts and understanding what they do.

Someone has offered to send me a wyzecam so if all goes well I'll have one to play with in a couple weeks :+1:

dariopb, 2017-12-08:
Thanks, I read the scripts and I think I got an understanding of what should happen. I was able to enable telnet and I'll play with them during the weekend.
(ps: if for some reason you don't receive that cam, let me know and I will send you one as well)

guarantee2, 2017-12-23:
Hello all.  I found this thread by doing some research on the same subject(s)  Does anyone want to help develop new scripts for Xiaomi DaFang camera to work on an English platform with an original app?  Get a hold of me at trailblazers102@gmail.com  I would like to develop this into a company and need partner(s) that can help with technical development while I bring in the product and advertisement.   I have some ideas that we can discuss.  Thank you.

codahq, 2018-02-01:
Did you receive your Wyze Cam yet? I'm wondering if this effort has been abandoned.

GWustenstein, 2018-02-19:
Been lurking for a while, but I had to say thank you all for the efforts involved in creating a recipe to reflash these Wyze beasts.

Uptime on these things is sometimes a challenge, but 48h of RTSP from them is still useful.  I look forward to the day we have a simple open environment to reload on these with apt or similar support, but until then, these are handy to fill in the blanks not covered by the hardwired IP cams

fmarkf, 2018-03-09:
@lokkju Could you please explain how to flash the ISO image(that you've posted earlier in this thread) onto Wyzecam? I played with the RTSP for some time and would like to revert my camera back to the original Wyzecam firmware. 

Thanks!

jbeck22, 2018-04-13:
I received my wyce cam v2 in the mail just the other day.  So far the camera works great using the wyze app and stock firmware.  I haven't been able to get the original fang hacks to work.  Hoping someone finds a solutions soon.

giuarb, 2019-04-15:
wondering if now that Wyze has released the beta of the firmware that supports RTSP, that beta firmware may work on xiaofang.

cryptobrian, 2019-04-25:
you can flash the latest wyzecam firmware (with RTSP) on a xiaofang.. but you can never get it past the scan QR code step.

[7 more replies]