samtap/fang-hacks #244: Mac 34

tsilout, 2017-11-01:
Hi there and I have to say sorry if there was already a thread I didn't see. I have 2 mac 34 xiaofang cameras with the Europe issue. Even though I downgrade them to the old firmware I never get to connect them fully with WiFi so the hack cannot work. What I mean is that I never have the blue stable light, it's always blinking. Is there a solution or I should just tosh them away? Please help.

samtap, 2017-11-01:
You should be able to pair them with mi home (hack applies only on old firmware, so downgrade first)

tsilout, 2017-11-01:
I downgraded and tried with my home 4.0.11 but no luck either

samtap, 2017-11-01:
If you can't pair I suggest contacting Xiaomi or reseller to ask for support. It's not related to fang-hacks.

cyberundertaker, 2017-12-07:
I had the same problem. I have a new model camera with reset button instead of a pinhole.
The model came with 3.0.3.56 firmware and on the first try i downloaded mihome app from play store and i got error that it can only be used in mainland china. After that i tried older version of mihome app 4.0.11...it worked but only on my wifi network. Next thing i did was apply fanghacks over sd card...rtsp stream was working great but also on my wifi network only. I tried setting port fowarding and dyndns on my router bu due to my provider i was unable to get port 554 open. I contacted them and found out that ports are locked on server side. They did something but after that i tried checking if the ports are open and they were still not opened. Then i removed sd card and did a reset of the camera and connected it to  play store version of mihome app. After successful connection i got the same error as before. Then i did a complaint as app instructed and sent an email with  image of qr code at the bottom of the camera and proof of purchase from ebay. Tried again with sdcard in camera and fanghacks and got rtsp withou any problem. At that point it was pretty late and i gave up for the day. I turned off my camera from power outlet with sdcard still inside. Next day after i got home from work late evening i removed the sdcard from camera and turned it on. I noticed that the light stopped blinking blue and was solid blue... I tried again connecting it with latest play store version of mihome app and the light was solid blue again and no more error message when tried viewing the camera stream. And now it also works over 3g/4g.
I don't know if it had something to do with the mail i sent them or the fact that the camera was unplugged from power for 24 hours after setting it up but now everything works on latest mihome app. Of course...app settings are mainland china, language default, and update pluggins in camera settings aare turned off.
Can somebody give it a try like this. Maybe it will work.