samtap/fang-hacks #248: New cam, cannot access $IP/cgi-bin/status

zaqwsx3, 2017-11-03:
Got a new cam yesterday (MAC is 34:CE:00:...).  No problems getting it working to the MiHome app and video works fine.  Unfortunately upgraded the firmware.  Tried fang-hack which installed fine (heard 2 chimes), however I could not access $IP/cgi-bin/status.  Confirmed $IP is correct via the app too.  Found out I had to downgrade the firmware, so downloaded [3.0.3.56 from here](https://www.youtube.com/watch?v=mpzPWYONWZA), installed it, and the camera now speaks in English.  Then installed fang-hack again, heard the 2 chimes, however still cannot access $IP/cgi-bin/status.  I'm also unable to ping $IP, nor can I see the device's MAC address via arp.  Any advice?

kslmgd, 2017-11-03:
After downgrade did u run mi app again to setup wifi and then install sd 2 beep and use http://ur ip/cgi-bin/status

zaqwsx3, 2017-11-04:
Yes. I've even uninstalled and reinstalled the mi app several times.

JasonCampey, 2017-11-04:
Me too on this one.. Hear the chimes when SD card inserted MAC:34:CE:XXXXX one with the QR code on the bottom.

JasonCampey, 2017-11-04:
just tried again and got it going.. Id suggest checking the firmware downgrade went through ok.

zaqwsx3, 2017-11-05:
Hmm, still no luck for me.  I've downgraded to the original [v3.0.3.56 firmware](https://mega.nz/#!GBghwbpY!btf0jxHPFTPtifqNJNMHXEiRd3H5DkUtOJYOq8QpgfQ) where you need to rename 0.elf to FIRMWARE_660R.bin, write it to the card, pop it in the camera, then hold the setup button for 10+ seconds whilst you power on the camera. So I think that's working as the chinese language is back.  I've also checked via the Mi home app that the current version showing is 3.0.3.56 ... so hopefully that is the all good.  Also checked the IP address is 192.168.0.149, and confirmed the camera is working in the app.

Then I install [fanghacks V0.2.0](https://github.com/samtap/fang-hacks/releases) and use [Win32DiskImager](https://sourceforge.net/projects/win32diskimager/) to write it to the sdcard, power on the camera, wait for it to connect to the network, then I pop the card in the camera, I hear the 2 dings, then try to access $IP/cgi-bin/status but get no response.

The strange thing is, I can see the device is working in the Mi home app, however I cannot ping it, nor can I see it on my DHCP client lease list, or find it via arp.  Maybe its because I'm using a 192.168.x.x network?  I really don't know what to try next.

JasonCampey, 2017-11-05:
I noticed the language changed to English when the firmware downgrade was successful.  I didnt have to do any changing names of files just downloaded the FIRMWARE_660R.bin file directly.. Didmt flash an image or anything just copied it directly to the SDcard restarted the camera with the card in and held reset for 10 seconds.

samtap, 2017-11-07:
@zaqwsx3 You can find out the IP it is using in mi home, don't remember exactly where but I think it's a menu item called 'network info'?
@JasonCampey The fw with english voice samples is obviously not original but works either way

pluskal, 2017-11-18:
Hi, I have exactly the same issue. I have used dd to copy 0.2 on SD card, booted camera, inserted card and http://<cam_ip>/cgi-bin/hello.cgi is empty and /cgi-bin/status has 404.  I know, that I have updated the firmware first time when I connected the cam to MI Home app. So, I have tried to downgrade it with all bins that I was able to find here, like 
[YouTube](https://www.youtube.com/watch?v=mpzPWYONWZA)
[Wiki](https://mega.nz/#!GBghwbpY!btf0jxHPFTPtifqNJNMHXEiRd3H5DkUtOJYOq8QpgfQ)

but never I was able to put the camera to update mode. I tried to replicate the process from YouTube video, same process... same time interval, but always the yellow led started to flash about 20-30s and not about 2 min like on the video. 

I have a version with 34:... mac, and QR code. 

Is it possible that newer cameras have this process blocked? 

SD card should be ok, because camera starts recording each time I put the card in.

Any tips very much appreciated! Thanks.

kuniyukikou, 2017-11-22:
@pluskal I got a 78 Mac XF from Taiwan. I can confirm that the downgrade method from handler is working, and no 404.

pixeldoge, 2018-01-10:
Hi,

I can downgrade to firmware 3.03.56 but had a hard time to hack the camera. The log shows below messages, I am wondering if the SDcard is partitioned correctly as I can't make vfat & ext2 partitions as per instruction as Windows keep saying it will not recognize the 2nd partition on removable media, so I just make 2 fat32 partitions instead. anyone can help? Thanks

Contents of /tmp/hacks.log:
Tue Oct 11 00:00:23 GMT 2016 - /etc/fang_hacks.sh: Executing script (enabled: 1)
Tue Oct 11 00:00:23 GMT 2016 - /etc/fang_hacks.sh: Failed to find hacks in /media/mmcblk0p2/data!
mount: mounting /dev/mmcblk0p2 on /media/mmcblk0p2 failed: No such file or directory
Tue Oct 11 00:00:23 GMT 2016 - /etc/fang_hacks.sh: Failed to find /media/mmcblk0p2/data!
Wed Jan 10 14:34:34 GMT 2018 - /media/mmcblk0p1/snx_autorun.sh: Running (device: mmcblk0p1)
Wed Jan 10 14:34:34 GMT 2018 - /media/mmcblk0p1/snx_autorun.sh: Starting telnetd