samtap/fang-hacks #250: 404 and no files on /media/mmcblk0p1

fdgago, 2017-11-05:
Hi.
Hi have problems with the first step. I receive the 404 error and on the helllo.bin I receive an empty page.

I have activated the telnet lines on the autorun script and I have accessed via telnet to the camera.This implies that the script has been executed.
The log file hacks.log seem to be correct.

The problem I have is that I don't have any files on the /media/mmcblk0p1 and mmcblk0p2 directory's. The ls command is empty.

What can be the problem?
I have created the image of the sd-card using Win32 Disk Imager.
Thanks

samtap, 2017-11-07:
Post firmware version and output of 'mount' command

fdgago, 2017-11-07:
The problem can be due a special kind of error in the Microsd or incompatibility.
I have changed the Microsd and it works.
Thanks for your support.

closed 2017-11-07 by fdgago