samtap/fang-hacks #252: RSTP not running (maybe i ran the scripts wrong)

masterchop, 2017-11-06:
this is the logs :+1: 
```
Contents of /tmp/hacks.log:
Sun Nov  5 21:28:15 GMT 2017 - /etc/fang_hacks.sh: Removed stale logfile
Sun Nov  5 21:28:16 GMT 2017 - /etc/fang_hacks.sh: Executing script (enabled: 1)
Sun Nov  5 21:28:16 GMT 2017 - /etc/fang_hacks.sh: Waiting for cloud apps...
Sun Nov  5 21:28:16 GMT 2017 - /etc/fang_hacks.sh: iCamera is running!
Sun Nov  5 21:28:16 GMT 2017 - /etc/fang_hacks.sh: Boa webserver is running!
Sun Nov  5 21:28:16 GMT 2017 - /etc/fang_hacks.sh: Not linking /media/mmcblk0p1/bootstrap/www/action: already exists
Sun Nov  5 21:28:16 GMT 2017 - /etc/fang_hacks.sh: Not linking /media/mmcblk0p1/bootstrap/www/func.cgi: already exists
Sun Nov  5 21:28:16 GMT 2017 - /etc/fang_hacks.sh: Not linking /media/mmcblk0p1/bootstrap/www/network: already exists
Sun Nov  5 21:28:16 GMT 2017 - /etc/fang_hacks.sh: Not linking /media/mmcblk0p1/bootstrap/www/scripts: already exists
Sun Nov  5 21:28:16 GMT 2017 - /etc/fang_hacks.sh: Not linking /media/mmcblk0p1/bootstrap/www/status: already exists
Sun Nov  5 21:28:16 GMT 2017 - /etc/fang_hacks.sh: Failed to find hacks in /media/mmcblk0p2/data!
Sun Nov  5 21:28:16 GMT 2017 - /etc/fang_hacks.sh: Failed to find /media/mmcblk0p2/data!
```

as you can see its talking about mmcblk0p1
on the instrucitons says you should go to 0. how to fix this?

as you can see it goes to

masterchop, 2017-11-07:
i found the test folder with the RSTP command but fails to find the libraries.

/media/mmcblk0p2/test # ./v4l2rtspserver 
./v4l2rtspserver: can't load library 'liblog4cpp.so.5'

libs are present user /mmcxxx0p2/usr/lib

i cant do softlinks to fix this

samtap, 2017-11-07:
There's something wrong with your sdcard. Try to flash the image again, or use a different card. The rtsp server is supposed to be controlled by a script. The binaries in test should not be used.

masterchop, 2017-11-12:
i see, i use the GIT repo, but some stuff were missing and collect them my self i guess i missed something.
i try the IMG File thanks.

closed 2017-11-12 by masterchop