samtap/fang-hacks #253: Anyone already try to connect camera to Hikvision NVR

solarij, 2017-11-06:
I hacked the camera and with TinyCam and IP Cam app on android work nice. I try to connect camera also to Hikvision NVR DS-7108N-SN through custom rtsp protocol, but until now without any success. Have anyone some idea how to do this?
In protocol menu are just three options (Auto, UDP and RTP over RTSP) and non are working.
[image]
[image]

samtap, 2017-11-06:
Does it expect RTSP stream over tcp or udp? http? Guess you'll need to ask Hikvision support.

masterchop, 2017-11-07:
How to start the rstp server , mine isnt starting

solarij, 2017-11-08:
Thanks for replay. Thats the problem with my understanding all of those protocols. I know that Hikvision support rtsp streams. I connect Axis camera with rtsp stream without problems. But here I don't know the different between "UDP" and "RTP Over RTSP". And there is no option "RTSP over TCP" protocol, like I used in android app ThinyCam.

samtap, 2017-11-08:
You could use i.e. vlc to connect with the Axis camera and check the logging output to determine the stream details. Or use netstat to check if it's using tcp or udp. Then make rtsp server on XiaoFang behave the same as the axis camera.

nevrending, 2017-12-15:
I've also tried this without success
My NVR is Hikvision DS-7608NI-E1

I have a XiaoYI Ants camera that works with the following settings

[image]

Tried to apply the same setting (with Stream Path changed to `/unicast`) didn't work
netstat-ing from the Fang, reveals that the NVR had connected to Fang's RTSP port

[image]

PS: Output stream from XiaoYI
[image]

dasmoover, 2018-01-05:
any progress on this?

can't connect via any vms sofware rtsp with
snx_rtsp_server -a -W 1920 -H 1080 -Q 10 -b 4096 -a

graemac, 2018-01-07:
Hello,

I am a bit of a newbie, please bare with me.  I am trying to enable sub stream for my cameras connected to my HikVision DS-7608NI-E2 configuration.  The aim being to get a URL (not rtsp) that I can paste into a browser or VLC to view an individual camera connected to the NVR.

Any ideas what the ‘stream path’ should be? 

My NVR has a local IP of 192.168.1.26

I have tried this syntax “ http://192.168.1.26/Streaming/channels/101/httppreview “ but getting the below error message.

Access Error: 404 -- Not Found

Can't open URL

Thank you in advance !!

ant-thomas, 2018-07-19:
Did anyone make any progress on this?
I'd like to add the 2 Fang hacked cameras I have to my NVR

S474N, 2018-10-28:
Have same problem with Hikvision NVR. Have also Xiaomi Ants and no problem, but with Dafang I cannot define working solution :(

Zippeyrude, 2018-11-04:
I have encountered the same problem.

I modded a Chuangmi camera successfully with [this guide](https://github.com/jymbob/chuangmi-720-hack), and I can see the RTSP feed in VLC with the stream path rtsp://IP_ADDRESS/live/ch00_0

But I cant get Hikvision to see it,

Zippeyrude, 2018-11-04:
> 
> 
> Have same problem with Hikvision NVR. Have also Xiaomi Ants and no problem, but with Dafang I cannot define working solution :(

Could you paste the exact config screens for the Xiaomi Ants, to see if others work?

Below are the screens I am using.

Does it make a difference if the config is on the browser or via the NVR console interface directly?

[image]

[image]