samtap/fang-hacks #255: Feature Request: Secure FTPd

mouth4war, 2017-11-08:
Would it be possible to replace the existing anonymous FTPd applet with a more secure one?

samtap, 2017-11-08:
For security you should not use ftp at all but ssh/sftp instead. The current ftpd is part of busybox so was easy to include. Of course it is possible to replace it, do you have a suggestion?

mouth4war, 2017-11-08:
Yes, simple fix to the ftpd script of adding "-u root" to the tcpsvd command line.

mouth4war, 2017-11-08:
However, this seems to not work for FTP. I can only connect via SFTP and not FTP.

FTP gives a control channel error.

samtap, 2017-11-08:
The -u option for tcpsvd has nothing to do with authentication

mouth4war, 2017-11-08:
Well it does seem to do something. When I start using that option, anonymous login for FTP isn't allowed anymore. It prompts for user and password. I get the control channel error after providing them though. Tried with and without PASV mode.

samtap, 2017-11-08:
Looks like it is supported in a newer version of busybox: https://git.busybox.net/busybox/tree/networking/ftpd.c#n39 but I still don't like it. It's the perfect way to leak your root password in clear text over the network, which can then be exploited for ssh etc.

mouth4war, 2017-11-08:
OK I found something. See https://github.com/shadow-1/yi-hack-v3/blob/master/src/busybox/.config

There is a flag in busybox config CONFIG_FEATURE_FTPD_AUTHENTICATION=y

mouth4war, 2017-11-08:
How about if it's configurable via the GUI and the user decides? Maybe a warning too?

samtap, 2017-11-08:
How about you build an updated busybox yourself and submit a pull-request? ;-)

mouth4war, 2017-11-08:
I would if I could. Teach a man to fish?

Also, an unsecure FTP already gives access to the video files on the camera so I would argue that a secured FTP (even with root) will be better than nothing.

samtap, 2017-11-08:
That's why you shouldn't really use FTP anymore. I only used it for convenience until I got dropbear to work and haven't used it since. If I provide an image with ftpd+auth, users will get the impression it's secure. It the same reason why I refused to accept the patch that 'hacks' authentication support in rtsp server. It only implies things are secure but they really are not.

But I agree it may be useful to some, and I've been planning to build a newer version of busybox anyway. So I will enable that config option when I get around to it.

closed 2018-03-05 by mouth4war