samtap/fang-hacks #256: Feature Request: Cycle Between Wi-Fi Options

nimdy2017, 2017-11-08:
Would it be possible to have an option to cycle between wireless client and access point, or do a check to see if the wireless client SSID exists and drop to access point if it doesn't?

I use my cam as a baby monitor so have it set to wireless client using my home SSID. I would like to be able to also take the cam away with me when we stay with relatives or on holiday. I know I can log into the FangHacks page to change the setting, but if I forget and simply pack it up when I go away, I'm left with a useless device whilst away. I wondered if a script could run that checks for my home SSID on boot. If it exists, connect to it, else it drops back to the access point settings. This would make it pretty fool-proof for my family to use. 

Thanks.

samtap, 2017-11-08:
If you know in advance you can just add the network in wpa_supplicant.conf (multiple networks are allowed) but I agree and planning to improve this in new version of the network scripts.

toshovski, 2017-11-11:
I use the camera the same way. I think it is a nice feature in case you are away.

Powerbless, 2018-01-14:
in my wpa_supplicant.conf the ssid and psk have only "xxxxxx" value. is the first line only a sample line? and can i replace the line or delete?