samtap/fang-hacks #263: What can i do without SDcard slot ?

canuyanik, 2017-11-16:
Hi,

I've realized that, my camera's sdcard slot not working. I've tried with different cards. But not ringing any sound when i put in. Not recording any video file. So i could not downgrade firmware. 

How can i access this device without regular sdcard methods ? outside of mainland china surely :)

samtap, 2017-11-16:
It's likely just the card-detect pin to the side of the sdcard slot is bent, easy to fix if you can take it apart and put it back together. Another way to access it is via serial console which you could then use to setup a NFS share with fang-hacks, but that's way more complex.

canuyanik, 2017-11-17:
Can i repair this bent on sdcard slot ? 
I can disassemble camera. But i have no idea whats going on inside micro sd card slot.

samtap, 2017-11-17:
Yep you can usually fix it