samtap/fang-hacks #265: seriel connection

kristianstage, 2017-11-18:
Hi.
I am stuck on the newest firmware. could not downgrade by sd card.
I opened a serial connection which connects fine.
I place 0.elf and other files on sd card as in wiki

I login with root / ismart12
i copied 0.elf to /tmp

but when i try to execute it with ./tmp/0.elf it say file not found. and is i cd into /tmp and ls I see the file is there.

Anyone know how to fix this. i tried with different sd cards as well nothing seems to work.
my mac is 34 xxx

samtap, 2017-11-21:
Is the file executable? (chmod 755 0.elf)

kristianstage, 2017-11-21:
yes i did that. but it still say file not found, even if I can see it with ls in /tmp folder :( hmmm

kristianstage, 2017-11-21:
if I just write ./0.elf in tmp folder i get " Segmentation fault"