samtap/fang-hacks #268: Adapting fang-hacks to Wyze firmware

aaronsilber, 2017-11-28:
This discussion formed out of issue #243, but is distinct as this issue is regarding getting fang-hacks on a Wyze Cam, and not Wyze firmware on a Xiaofeng.

Several users have expressed interest in adapting fang-hacks in one way or another to run on Wyze Cams. There has already been effort from @samtap, @ril3y, et al. in #243 to pursue this.

Personally I am very interested in getting raw video streams out of these devices and disabling the ThroughTek P2P functionality. There are many ways the Wyze firmware could be improved upon that would afford more options for interfacing with the unit.

dustinsterk, 2017-11-28:
Also adding this thread as Wyze is investigating on UDP traffic to non North American Servers:  https://www.reddit.com/r/wyzecam/comments/7cykgf/wyzecam_sending_data_to_servers_other_than_aws/

ril3y, 2017-11-29:
I wrote a Dockerfile tonight to get the toolchain all setup for anyone who wants to play with it.
https://hub.docker.com/r/ril3y/sonix-sn98600-toolchain/

I have the wifi configuration working without the app so far.  A few other things up my sleeve but not working yet.

dustinsterk, 2017-11-29:
Was having an issue pulling it....had to issue the following:

docker pull ril3y/sonix-sn98600-toolchain:firsttry

ril3y, 2017-11-29:
Yah kiteamatic does not like the tag i gave it.  Try it now should work
with latest.

Riley

dustinsterk, 2017-11-29:
Yep..good now, thanks.  To confirm, are you building this into a img file and then flashing it to the camera to test on the device?  If so, how are you flashing?

ril3y, 2017-11-29:
Have not got that far.  For the life of me I cannot figure it out.  I had it building last night the kernel / fs everything in my docker image.  go to buildscripts `type make config` it will ask you a bunch of questions which some of I admit is a guess.  But today its failing so I am backtracking trying to figure out what I messed up.

dustinsterk, 2017-11-29:
Ran through with the defaults and it created the config file just fine for me.  Is it failing for you creating the config or during the build?

dustinsterk, 2017-11-29:
When I build I am seeing:
arm-linux-ld --host=arm-unknown-linux-uclibcgnueabi   -r -o libstubs.o  stubs.o
arm-linux-ld: unrecognized option '--host=arm-unknown-linux-uclibcgnueabi'
arm-linux-ld: use the --help option for usage information
make[3]: *** [libstubs.o] Error 1
make[2]: *** [examples/standalone] Error 2
make[1]: *** [uboot] Error 2
make: *** [u-boot-2011-09] Error 2

ril3y, 2017-11-29:
Lets not hijack this thread.  I guess we can open an issue sorry pat!  on
the cross compiler.

spikebike, 2017-12-01:
Has anyone tried a normal fang-hanks install on to a WyzeCam?  What exactly happens?

joeand37, 2017-12-13:
@spikebike The WyzeCam ignores the script and never loads. I was able to install a standard Xiamoi Firmware then utilize Fang-Hacks. My results may not help the core of the thread but certainly answers your question.  ;-)

jat255, 2017-12-13:
@joeand37 could you provide some simple instructions on that process? Is there anything different you had to do than described in the [readme](https://github.com/samtap/fang-hacks#general-usage)?

JohnOKeefe15, 2017-12-18:
@joeand37 Was there anything special you needed to do to be able to load the standard Xiamoi firmware on the Wyzecam? Which Xiomoi firware version did you use on the Wyzecam? If possible could you post a short list of steps you did to do so? If what you are describing works then we should be able to put custom firmware on any Wyzecam. Thank you in advance for any help provided!

ril3y, 2017-12-18:
I am not sure how this works.  The binary is "signed" or something and when
you try to take a Fang bin and flash it, it fails to flash (watching the
serial output).

Riley

samtap, 2017-12-24:
I tried to intercept the Wyze firmware update, but it uses HTTPS! So not much fun with wireshark, I'll probably have to get it off the device itself :(

dasmoover, 2017-12-28:
@samtap https://hack-ed.net/2016/03/31/introduction-to-mitm-with-sslstrip/

sslstrip

@joeand37 please release a guide. rtsp would be amazing.

joeand37, 2017-12-29:
This is on my list and I promise I will write one soon. I had to use an amalgamation of different sources to get it to work and I just need to compile them.

dasmoover, 2017-12-31:
thank you @joeand37 
I look forward to it!

joeand37, 2018-01-04:
# You will not be able to use the WyzeCam app after performing these steps!

## Download the following items to start:
1.	Factory Xiaomi firmware: https://github.com/samtap/fang-hacks/wiki/HowTo:-Flash-original-Xiaomi-firmware-from-sdcard-(factory-reset)
2.	English firmware (Use after factory "Chinese" firmware has been flashed): https://www.youtube.com/watch?v=mpzPWYONWZA -- BIN file Download located in the description of the video (sorry Publisher doesn’t want reposting of DL link).
3.	Fang Hacks Image: https://github.com/samtap/fang-hacks

## Prep the SD Card
Create the fang-hacks SD Card per the instructions on Github (I used the Win32 Disk Imager method found [here](https://github.com/samtap/fang-hacks/releases)).

Once the SDCard is ready, browse the card from your PC and rename the “snx_autorun.sh” to something like “snx_autorun.xx”. This step prevents the fang-hacks being executed instead of the firmware (I don’t know if it is needed, but seemed like a good idea). Now extract the 0.elf file to the SD Card and rename to “FIRMWARE_660R.bin” (sans quotes).

Now follow the Video on how to flash the camera.

## Summary of steps:
### Base firmware load:
1.	Hold down the setup button for 10 seconds BEFORE connecting power
2.	With button still depressed, plug in power to the camera and continue to hold for another 10 -12 seconds and wait until the connection light blinks continually.
3.	Press setup button and you should hear a Chinese voice prompt if successful. (if you understand mandarin then you can skip the English steps).

### English Firmware (Chinese firmware required before proceeding):
1.	Remove the SD Card from the camera and connect to PC
2.	Replace the “FIRMWARE_660R.bin” with the version downloaded from Yk Handler’s Youtube video.
3.	Remove the SD Card from the PC and repeat steps 1 – 3 from the base firmware instructions
4.	If you hear English prompts it was successful
 
## Setting up WiFi in Mi Home app:
You must configure the camera with the MI Home app. I used version 4.0.8 for Android as this seemed to have the best luck at configuring the wireless. Make sure to turn off auto updates in the app. Secondly, It may prompt to update the camera firmware, make sure you don’t accept.

I found 4.0.8 on [apkpure ](https://apkpure.com/mihome/com.xiaomi.smarthome/download/60614-APK?from=details%2Fversion)here.

Follow the prompts in the app and add the details specific to your WiFi. It will go through the network connection, and you should hear the connection successful message from the camera. However, the app will fail to add the camera to Xiaomi servers in the app (which is ok). 
Since the application fails to connect you will not know the IP address of the camera and the MAC address on the camera’s label is no longer valid. I had to look at my routers DHCP log to find the IP that was assigned. I also added a DHCP reservation so it wouldn’t change in the future.

## Loading fang hacks:
1.	Remove the SD Card and connect to a PC so you can rename the “snx_autorun.xx” back to “snx_autorun.sh” and remove the “FIRMWARE_660R.bin”. 
2.	Safely remove the SD Card from the PC and place it back into the camera.
3.	Once you hear the chimes, within 1 -2 minutes you should be able to load the Fang-Hacks page (HTTP://<CamIP>/cgi-bin/status)

### All done!
If you are successful at getting Fang-Hacks to work, I suggest running through the Bobby Romeo’s [guide ](http://bobbyromeo.com/technology/xiaomi-smart-1080p-wifi-ip-camera-rtsp-streaming-hack/) to optimize the camera and fix the Heat Issue caused by the IR Control script.
Note: I had issues with my 32GB SD cards loading the firmware, and had to use a 64GB to flash the Firmware ONLY. I was then able to to use the 32GB cards to load Fang-Hacks and keep in the camera.

ril3y, 2018-01-04:
>    fang-hacks/wiki/HowTo:-Flash-original-Xiaomi-firmware-from-
>    sdcard-(factory-reset)
>    load the Fang-Hacks page (HTTP:///cgi-bin/status)

You can avoid all of the APK / mi home stuff by using my qr code generator
code pen.

JohnOKeefe15, 2018-01-04:
@ril3y Do you have a link to your QR code generator? Also can you provide the steps that one would follow if they skip the APK/mi home steps and use your qr code generator? Thanks!

ril3y, 2018-01-04:
https://codepen.io/ril3y/pen/gXyzmO

joeand37, 2018-01-04:
@JohnOKeefe15 You can skip all but the last sentence of "Setting up WiFi in Mi Home app".
Instead, use the code generator to set up the camera.

JohnOKeefe15, 2018-01-04:
@joeand37 OK, thanks. I got everything going. I noticed that the fang hacks version is fanghacks_v0.2.0.zip. Is that the latest version? Also is it possible to go back to the original WyzeCam firmware? It is not a big deal if I can't but just helpful to know if there is a process to reverse this. Thanks again for writing up the steps and the help!

joeand37, 2018-01-04:
@JohnOKeefe15 , try flashing the English firmware. This way you can ensure it is giving you the correct responses. I did have a similar problem when I tried to skip the English step 
Lastly you can try a power cycle then check your router for the most recent ip allocation.

JohnOKeefe15, 2018-01-04:
@joeand37 Using the English firmware got me farther. This time when I pointed the camera at the QR code it said it received the info and then it went to flashing blue light. When I looked at my router I found an IP address for the device. Next I powered off the device and removed the SD card. I renamed the “snx_autorun.xx” back to “snx_autorun.sh” and removed the “FIRMWARE_660R.bin”. When I booted it connects to the wi-fi network but it just stays in a flashing blue light state. It never goes to solid blue or plays any chimes. When I try to navigate to (HTTP://MYDEVICEIP/cgi-bin/status) it says The requested URL /cgi-bin/status was not found on this server. After I scanned the QR code was I supposed to wait until the device got to a solid blue state?

JohnOKeefe15, 2018-01-04:
@joeand37 Looking through the comments of the Youtube video of the English firmware some other people say they get a continious flashing blue light. Also one comment talks about navigating to: http://MYDEVICEIP/cgi-bin/hello.cgi. I went to that address and it returned:
snx_autorun.sh
bootstrap
System Volume Information
time_lapse
record

So it looks like I can connect to the camera over the network. It just seems that the status is not working. Any thoughts on where to go from here?

dasmoover, 2018-01-05:
Thank you. Managed to get it working.

Unfortunately the native rtsp implementation doesn't work on most popular VMS systems.

Does not work with Hikvision, Dahua or Exacq rtsp implementations.

joeand37, 2018-01-05:
I use Blue Iris and it works perfectly. I also tried iSpy and it worked too.
You could use ffmpeg server or VLC as an intermediary between your VMS and the camera, I know it's not ideal but it works. 
Spending $20 per camera vs 100+ may be worth the annoyance.

ibnharoon, 2018-01-09:
Sorry dumb question: "What is VMS?".

joeand37, 2018-01-09:
Video Management Software.

ibnharoon, 2018-01-09:
Do you have the original firmware just in case I brick the camera?

ibnharoon, 2018-01-09:
Just to get this right, do I have to flash "Chinese" firmware before "English" firmware?

openipcamera, 2018-02-08:
https://openip.cam

official fork for wyze

samtap, 2018-02-08:
@openipcamera Good job at stealing my work and trying to make a buck off of it selling pre-flashed sdcards. If your claims are true, please share the changes responsible for making it work on all firmware versions of Xiaomi/Wyze cams, as dicated by license terms.

openipcamera, 2018-02-08:
@samtap 
Hello, I did not intend to steal any of your work I have given you credit on all the readme and documentation. OpenIPC is a fork of your work adapted for use with the Wyzecam. Outside of a few minor modifications (wifi/remove rtsp audio) I have included a flashable firmware in the .img file that works with Wyzecam and Fanghacks. Everything is accessible from the repos in my account. I have also removed all prepaid sdcard options and will leave the project truly FOSS. Thanks for your time.

jat255, 2018-02-08:
@samtap @openipcamera not to stick my nose in anything (I'm not an IP lawyer, so this may be completely wrong), but I think under the the CC BY-SA 3.0, there's nothing prohibiting the sale of something like this.  A little shady, perhaps, but probably not illegal. 

From what I can tell though, there's no mention of a license on @openipcamera's repository, nor is there an indication of what changes have been made, so those would appear to be violations of the fang-hacks license. Also, calling it an "official fork for wyze" is probably also a violation, since you cannot give attribution "in any way that suggests the licensor endorses you or your use." Again, IANAL, but I like trying to figure this sort of stuff out, for some reason.

samtap, 2018-02-08:
There's a note in the README.md and that's fine with me. Even selling sdcards is fine with me as long as I don't get any support requests ;-).

But really this thing can do so much more, if a couple of devs could combine efforts, which is why I put stuff on github in the first place. And as far as I know, there're issues with downgrading latest firmware so if there's a solution for that it should be shared publicly.

openipcamera, 2018-02-08:
@samtap I understand and agree. We are looking to branch out to other large manufacturers eventually. Let me be more clear: we have not made any patches to fang-hacks, rather we found a solution to load a firmware that can receive the fang-hacks payload. The stock Ziaomi firmware can be flashed to the Wyze latest firmware and it takes the fang-hacks payload.

spazfishy, 2018-02-12:
Forgive me for being a lil uneducated on this but I have a question. Once you apply all of these hacks how do you go about getting the username and password to use this camera on an app such as blue iris or similar?

openipcamera, 2018-02-12:
@spazfishy there is no user/pass set by default.

rtsp://deviceip/unicast

openipcamera, 2018-02-12:
@samtap it seems the firmware bin file in our fork can downgrade any camera regardless of firmware.

Blue-Sauce, 2018-03-15:
Wyze just released the .bin files for their v3.9.1.102 and v3.9.2.30 firmware https://www.wyzecam.com/forums/topic/flashing-your-wyzecam-v1-firmware/

AdrianTP, 2018-04-11:
I am curious about whether any similar work is being done for the v2 -- they claim that they put completely custom hardware inside, but I highly doubt that, and it would be awesome if I could get this using RTSP so I don't have to rely on Someone Else's Computer to keep my privacy private.

JohnOKeefe15, 2018-04-11:
I am curious about the same thing for v2. I guess someone is going to have to just try or provide one to one of the main developers.

AdrianTP, 2018-04-17:
Thank you for the link, @samtap, and for all your hard work on the v1. You've given me something to aspire to. :)