samtap/fang-hacks #269: Possibility of using Python as a scripting language

afeezaziz, 2017-11-28:
Hi,

I have been using this hack to stream images but I am wondering whether it is possible to use Python to run a  few of my scripts?

samtap, 2017-11-29:
Sure it is possible, just compile it and put it on the sdcard. Note that many of the more interesting python modules depend on C/C++ libraries that need to be cross-compiled, you can't simply run 'pip install whatever' on an embedded arm platform. And regular python will not perform very well so I would start with MicroPython and doing some tests to get a rough idea of the performance.

puz12zek, 2018-02-27:
@samtap Can you please elaborate more on the process for compiling Python for ARM?
I have tried to copy a compiled version from a docker image, but that didn't work...

russkel, 2018-03-13:
To add to the list the linux version is also quite ancient.