samtap/fang-hacks #270: IR image quality

350d, 2017-11-29:
Is this a normal quality for IR mode from this camera?

[image]

Log from the camera: [hacks.log](https://github.com/samtap/fang-hacks/files/1512516/hacks.log)

Also, stream in VLC freeze and hangs constantly. WIFI router is like 5 meters away from the camera.

samtap, 2017-11-29:
From the snapshot it looks like the IR leds are not turned on

350d, 2017-11-30:
@samtap but leds are turned on and I can hear that exact "click" sound and see red lights on it.

TommyChausson, 2017-11-30:
Hi,

Is'nt it the same problem here :  https://github.com/samtap/fang-hacks/issues/182 or here https://github.com/samtap/fang-hacks/issues/61 ?

it seem's that there is no solution for the moment  .

samtap, 2017-11-30:
Take a look at what the [script](https://github.com/samtap/fang-hacks/blob/0d2913f11b6bcf471f97e7bb29e6d4b5777c1dbf/usr/bin/fang-ir-control.sh) does and run some commands manually? I'm not affected by the issues reported so no idea what I can do to fix it....