samtap/fang-hacks #271: hack for the new version xiaofang?

fatfredycat, 2017-11-29:
Do you plan to make a new hack for the latest version? Many of us would be grateful!
Latest version is 3.25?Thanks

samtap, 2017-11-29:
Please post exact version string and in what way it doesn't work. Note you'll need to downgrade first to apply the hack.

fatfredycat, 2017-11-29:
I have tried to downgrade from sd but it does not flash
it seems to me that the exact version is 3.2.5.0 or 3.2.0.5

samtap, 2017-12-03:
Are you sure you're talking about a Xiaomi XiaoFang (white cube camera, gray magnetic base)? 

I've checked one of my cams in mi home today, latest firmware version is still 3.2.0.30, I can't upgrade to anything newer.

fatfredycat, 2017-12-04:
Okey...the version is 3.2.0.30...can I downgrade with sd metod?ot not working in this version...

samtap, 2017-12-04:
You can downgrade using uboot if you put FIRMWARE_660R.bin in fat32 partition on sdcard. Discussed in detail on wiki and other issues so closing this.

closed 2017-12-04 by samtap

fatfredycat, 2017-12-04:
In cam with 34:x:x:x mac?

samtap, 2017-12-04:
MAC has no relation to firmware updates