samtap/fang-hacks #272: Partition expand reboot my XiaoFang camera is bricked

balintbabics, 2017-11-29:
I flashed the release img to my sd card with Win32 imager and after the first start the `/status` page said it has to expand the second partition (ext2), I pressed to execute it, successfully finished and said it needs a reboot to apply, so I hit the Reboot button.

After that my camera won't come to life, only the orange LED lights, constantly right after when I plung into AC.

I tried to reflash the firmware (Setup button pressed while powering up) but it's not worked.

So what should have I do to bring back my camera?

drewscott, 2017-12-02:
From the README: "Currently there's no known recovery methode, i.e. an image to flash to the device when it doesn't boot, though you can solder wires to it and get access to a serial console.  ........ As long as you don't completely disable the cloud apps (DISABLE_CLOUD=1), you can always revert to original behavior"

samtap, 2017-12-03:
You'll need the recovery procedure, put fang_hacks_rescue.cfg in bootstrap folder and reboot twice.

balintbabics, 2022-12-10:
Sorry, I've missed coming back here. I managed to bring back my camera with the method above, so this issue could be closed.