samtap/fang-hacks #273: Telnet/SSH password not working

tcmoreno, 2017-12-03:
Hi,
I have been able to install without issues the fang hacks and using my camera with RTSP. Version is 3.0.3.56

However I cannot access the camera via SSH/Telnet. I get password error (root/ismart12). On the other side FTP works perfectly.
Any idea of what can be happening?

samtap, 2017-12-03:
If you updated to latest version possibly the root password is changed. Happened to one of my cams (but not on others, no idea why). Can you run commands using the snx_autorun.sh script? You could put a passwd command in there to change it.

tcmoreno, 2017-12-06:
I will try that way, should look like this?
# echo "password" | passwd --stdin root

samtap, 2017-12-06:
Use this: `echo -e "password\npassword" | passwd`
I'm curious what the changed passwd is, so could you add something like `cp /etc/shadow /media/mmcblk0p1` before that, to copy the passwd hash file to sdcard?

ril3y, 2017-12-06:
Can you get the new iSCS binary file too once on?

tcmoreno, 2017-12-06:
Not working yet, my snx_autorun.sh file looks like this

`#!/bin/sh

[ -z "$MDEV" ] && return

cp /etc/shadow /media/mmcblk0p1
echo -e "password\npassword" | passwd

echo "$(date) - $0: Running (device: $MDEV)" >> /tmp/hacks.log 2>&1

# if [ ! $(pidof telnetd >/dev/null) ]; then
#   echo "$(date) - $0: Starting telnetd" >> /tmp/hacks.log 2>&1
#   telnetd &
# else
#   echo "$(date) - $0: Telnetd already running! (PID: $(pidof telnetd))" >> /tmp/hacks.log 2>&1
# fi

if [ ! -d "/tmp/www/cgi-bin" ]; then
	mkdir -p "/tmp/www/cgi-bin";
	echo "$(date) - $0: Created /tmp/www/cgi-bin" >> /tmp/hacks.log 2>&1
fi
for i in /media/$MDEV/bootstrap/www/*; do
  echo "$i -> /tmp/www/cgi-bin/$(basename $i)"
  ln -sf "$i" "/tmp/www/cgi-bin/$(basename $i)" >> /tmp/hacks.log 2>&1
done 
`

samtap, 2017-12-06:
"Not working" doesn't tell me anything. Does it not run the script? Is the shadow file copied? Does it output any errors? Append `>> /tmp/hacks.log 2>&1` to redirect output to log and `cp /tmp/hacks.log /media/$MDEV` to copy log to sdcard. Uncomment other stuff not needed. You need to be on older fw as the new fw versions don't run snx_autorun.sh

lanhed, 2018-01-10:
Any luck extracting that password yet? I have the same setup and can't use ssh.

tcmoreno, 2018-01-12:
Nope, I gave it up :-(
Good luck with it...

icuye12, 2018-12-16:
Hi ,I have the same problem. But I deleted the file in /etc/init.d/videomdprob.sh. so who can get me the videomdprob.sh file.
I can not to login the system.by (root/ismart12).