samtap/fang-hacks #274: Xiaofang Brick

mccanos, 2017-12-04:
Hello to all the community and thanks a lot for the big support of this community,

i have made the fanghacks to my xiaofang but I have had a problem after expand the memory and now my xiaofang is bricked. I have tried to recover it with the fang_hacks_rescue.cfg but it has not worked for me. i have also tried to reflash the firmware without success so I write to ask help.

I have access to the serial port and I have soldered the wires and I can see the camera with Putty in my laptop. I have tried to flash the firmware via serial port but the problem is that my camera doesn't mount the SD card and I can't do it. I think that I have the fang hacks installed because I have an error in relation with it. This is the dtaa that I see in Putty regarding this last point:

starting pid 529, tty '': '/etc/init.d/rcS'
/etc/init.d/rcS: source: line 2: can't open '/etc/fang_hacks.cfg'
starting pid 531, tty '/dev/ttyS0': '/sbin/getty -L ttyS0 115200 vt100'
usb 1-1: new high speed USB device using snx_ehci and address 2
usb 1-1: New USB device found, idVendor=0bda, idProduct=0179
usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
usb 1-1: Product: 802.11n NIC
usb 1-1: Manufacturer: Realtek
usb 1-1: SerialNumber: 34CE00D4A06E

If I have the fang hacks maybe I could do something via serial port to force to change the network mode and disable cloud in order to try to recover the camera but I need some help to know what files could I test to recover the problem as at this moment I don't have clear following steps....

Thanks a lot in advance and best regards,

samtap, 2017-12-04:
Did you try with a different SD card? If you have serial access you can also just fix it manually by editing files in /etc

mccanos, 2017-12-04:
No, I've only tried with one SD Card but previously to apply the fang hacks I've been able to downgrade the firmware with this SD card and I have also able to apply the fang hacks with the same SD card si I think that the SD card is not the problem. I'll try to get other SD card and test it.

The main problem is that when I send the command ls to see the main files I don't have the etc folder. I only have one folder in the camera but it's not the etc folder. I don't remember now the name but I'm sure that it's not the etc folder so it's possible that some archives are corrupt but I'm not sure how can I recover or rewrite it.

samtap, 2017-12-04:
When the filesystem resize doesn't finish the card is likely left in a corrupted state. You'll need to write the image again using etcher or win32 disk imager etc.

mccanos, 2017-12-07:
I've tried to write the image again using win32 disk imager but I have had same result. I think that one problem is that the SD Card is not automounted. I have tried to switch on with the SD card and also to introduce the SD card after switch on without success. In both cases I have same message:

/etc/init.d/rcS: source: line 2: can't open '/etc/fang_hacks.cfg'

When I check the files in the camera with putty, I only have the etc_default folder and inside it I have several documents:

TZ                   inetd.conf           profile
TZ~                  init.d               protocols
Wireless             inittab              rc
ds_font_table        mdev.conf            resolv.conf
firmware             miio                 services
fstab                mtab                 shadow
group                network              syslog.conf
hosts                no_update_list       twowayaudio.conf
hotplug              passwd               wpa_supplicant.conf

Maybe would be possible to modify one of this documents to recognize the SD card? Or maybe would be possible to be able to connect to MI cloud and restore the firmware?

Thanks in advance and best regards,

samtap, 2017-12-07:
I don't understand, you say the camera is bricked but you can still connect with putty?? The files you describe are in /root/etc_default.

mccanos, 2017-12-07:
Really it's not completely bricked because as you say I can connect through putty. The problem is that the orange led is always orange (fixed, not blink) and it not recognize the sd card so I can't try to rescue it or reflash the firmware.  Maybe if I can modify something in root archives to automount the sd card I could recover it but I need help with this last part to make the modifications.  Please could you help me with this part?

Thanks a lot in advance for your support to the community

samtap, 2017-12-07:
Did you try mounting the sdcard manually? Also show us the contents of /etc/os-release

mccanos, 2017-12-07:
I connect with putty through serial port.  I can't connect through telnet or ssh, only by serial port.  Please could you explain me how can I mount manually the SD card? It's needed to modify something in the SD card or could I do it through putty via serial connection? Thanks again for your support.

renzjn, 2018-03-29:
I'm having similar problem. tried multiple tutorials. seems that firmware is broken. if i tried to boot the device it just have steady yellow orange light. and if i tried to connect to my phone by pressing the setup button it will just prompt a Chinese voice and led turns off. I  noticed that the camera is turned on coz of the 2 red steady LED light in front beside the camera lens. Anybody got similar problem and solution? thanks

renzjn, 2018-03-29:
sometimes the voice won't finish and immediately turn of the back led yellow light.