samtap/fang-hacks #275: Setup without Mi Home (i.e. network setup) - possible?

Mazo, 2017-12-04:
Is it actually possible to set up a Xiaofang without having to resort to doing the initial setup with the Mi Home application? Would certainly be nice if you could just edit a few configuration files to set up networking and then not have to even look at the Mi Home app.

Does the Mi Home app actually do any configuration on the camera beyond initially setting networking setitngs?

ril3y, 2017-12-04:
Yes,  I created a script that can configure the network without the app.
Msg me directly if you want to test it out.  I have not released it yet.

Riley

Mazo, 2017-12-04:
@ril3y Would love to test it, but didn't github remove private messaging years ago?

ril3y, 2017-12-04:
https://stackoverflow.com/a/44229207

Ha that worked on me but not on your account.

Riley

bomellberg, 2017-12-04:
I second the wish to try this. Just got 4 cams home that needs configuring. I would love to not have to use the app at all.

Mazo, 2017-12-04:
@ril3y Have sent an email to what I presume is your email address from the api.

Mazo, 2017-12-05:
Looks like someone has already worked this out actually, pretty simple following the instructions in #132

@ril3y the QR code setup would still be a really helpful thing for a lot of users though!

ril3y, 2017-12-06:
https://codepen.io/ril3y/full/gXyzmO/

bomellberg, 2017-12-08:
I can confirm that the QR generator is working. I just setup a camera in under two minutes. Start cam, press button, scan code, wait 20 sec, insert card, apply hacks, set timezone. Done!

I use motion + motioneye for supervision. Works great!

/Bo

Mazo, 2017-12-17:
This issue can now be closed. I've added a wiki entry with instructions on both methods for setting WiFi details without Mi Home.

https://github.com/samtap/fang-hacks/wiki/Connecting-to-WiFi-without-using-Mi-Home