samtap/fang-hacks #278: Successful downgrad + hack // Bootloop

bderenzo, 2017-12-08:
Hi,

I made a downgrad sucessfully.
Then I apply the hack, successfully and try to expand the fs. 

Now my camera boot until the fixed blue light. I can acces to fang-hacks but i have an error message : "Nothing is mounted on /media/mmcblk0p2!".
[image]
If a click on manage script the page is empty

I have tried to boot with or without sdcard, nothing work.
I can start and acces to the rescue telned but don't know what to do.

Is there something to do ?
Thanks in advance.

samtap, 2017-12-09:
You probably corrupted the filesystem, try to flash the image again.

bderenzo, 2017-12-09:
@samtap Thank you very much. It now works !
In fact its the resize2fs (approximately 1 hour) which cause the problem. I think I reboot during the operation.

closed 2017-12-09 by bderenzo

pixeldoge, 2018-01-10:
Hi, 

I had a hard time to hack the camera. The log shows below messages, I am wondering if the SDcard is partitioned correctly as I can't make vfat & ext2 partitions as per instruction as Windows keep saying it will not recognize the 2nd partition on removable media, so I  just make 2 fat32 partitions instead. anyone can help?  Thanks

Contents of /tmp/hacks.log:
Tue Oct 11 00:00:23 GMT 2016 - /etc/fang_hacks.sh: Executing script (enabled: 1)
Tue Oct 11 00:00:23 GMT 2016 - /etc/fang_hacks.sh: Failed to find hacks in /media/mmcblk0p2/data!
mount: mounting /dev/mmcblk0p2 on /media/mmcblk0p2 failed: No such file or directory
Tue Oct 11 00:00:23 GMT 2016 - /etc/fang_hacks.sh: Failed to find /media/mmcblk0p2/data!
Wed Jan 10 14:34:34 GMT 2018 - /media/mmcblk0p1/snx_autorun.sh: Running (device: mmcblk0p1)
Wed Jan 10 14:34:34 GMT 2018 - /media/mmcblk0p1/snx_autorun.sh: Starting telnetd